ARMmbed/DAPLink #94: If USB reset occurs in the middle of SCSI command mass storage stops working

c1728p9, 2016-04-20:
The USB stack does not propagate a reset to the endpoints.  Because of this if a USB bus reset occurs in the middle of a SCSI transaction the device will stop responding to mass storage commands.  This persists until the board is reset.

sukidog, 2016-04-20:
Curious. Under what circumstances does a USB reset occur in normal use cases? Or is this something the automated tests surfaced?

c1728p9, 2016-04-20:
This occurred after a couple of days of running automated tests every hour.  This shouldn't happen under normal usb operation.

VeliMattiLahtela, 2016-05-18:
We might have faced a similar situation during one of our automated runs:

What kind of symptoms are visible from the systems point of view?

Currently we have one device that is visible, and the serial interface can be accessed but the mount point is missing.

From dmesg we noticed this:

```
[18328.585149] VFS: busy inodes on changed media or resized disk sdc
[18328.617387] FAT-fs (sdc): unable to read boot sector to mark fs as dirty
[18328.620795] sdc: detected capacity change from 8454144 to 0
[18330.631344] sd 7:0:0:0: [sdc] 16512 512-byte logical blocks: (8.45 MB/8.06 MiB)
```

And sdc does not have a mounting point on the system.

closed 2016-07-10 by c1728p9