ARMmbed/DAPLink #100: Request for K64F DAPLINK reset without reset to the MCU.

VeliMattiLahtela, 2016-05-10:
Some of the cases where the K64F board has totally frocen from the system point of view, it would be nice to inspect is it DAPLINK that has frocen or the application running on the MCU.

c1728p9, 2016-05-11:
Hi @VeliMattiLahtela, can you provide more details on this?  Are you wanting DAPLink to reboot without resetting the K64F, or are you wanting a way to reset DAPLink?

One way you can narrow down the cause of a freeze is to try out the other endpoints.  If DAPLink is still running you should still have a filesystem and/or a functioning serial port and/or a HID endpoint.  Further, you could use a debugger like uVision or pyOCD and if you are able to connect then you can both rule out a DAPLink crash and debug the K64F.

VeliMattiLahtela, 2016-05-12:
The initial request would be to have a way to reboot DAPLINK without resetting the K64F. 
This case usually pops up during automation runs, unfortunately they are run with raspbian and at least pyOCD does not function well on those setups. On some occasions the serial can be seen on the system, but the mount point is missing. accessing the serial interface does not give any prompt, with the binaries we flash the serial will have prompt.

c1728p9, 2016-07-10:
Hi @VeliMattiLahtela, on the FRDM-K64F the power supply to the K64F is controlled by the k20dx, so if it is reset then the K64F will also be reset.  Aside from that, as part of the K64F init sequence DAPLink resets it.  You could use another board like the NRF51-DK or microbit, which does not reset the target on boot.  To get DAPLink to reset you could switch from interface mode, to bootloader mode and then back to interface mode.  Switching between modes in DAPLink causes a software reset.

c1728p9, 2016-07-12:
Closing this issue since DAPLink already has this ability, but can't on the K64F due to hardware limitations.  Feel free to re-open if you would like to discuss this further.

closed 2016-07-12 by c1728p9