ARMmbed/DAPLink #101: fea-request: flash sha1 to text file

jupe, 2016-05-10:
I would like to see afterward what binary was flashed to the device. Build sha1 could be good unique value for this. This feature would be very useful for CI/automation.

e.g. Use case:

1. user#1 flash binary to the device
   -> daplink write binary sha1 to some text file (e.g. DETAILS.TXT)
2. user#1 unplug device and give device to user#2
3. user#2 plug device
4. user#2 want to verify what binary was flashed
   -> read text file and check sha1

bridadan, 2016-05-10:
@PrzemekWirkus And I have also discussed this, would be really useful for determining when the device is fully flashed as well! This coupled with a "flash count" could be really helpful to be sure all file operations are completed and the target is ready to be reset. The "flash count" wouldn't have to be persistent across power cycles, but it'd be a great indicator in addition to the drive remounting that the flashing process is complete.

PrzemekWirkus, 2016-05-10:
There were certain features similiar to this discussed between me and Russ.
For example volatile and non volatile counters available via DETAILS.TXT:
* (volatile) Resets since last power cycle.
* (nonvolatile) Total reset/power on cycles.

Hard to say if SHA can be provided in bare-metal system (maybe too expensive to have this algorithm inside DAPlink f/w).

I think we just should know DAPlinjk's roadmap, it would be easier to know what is there and maybe give ideas what automation wants.

sg-, 2016-05-10:
Sounds like a good feature but need to make sure we don't add too much magic to the filesystem that other DAPLink like builds wont offer. We can add all sorts of features but need to really understand what is a requirement and how best to implement it such that all mbed-like boards can achieve the same status.

hugovincent, 2016-05-10:
Can I suggest SHA2 instead of SHA1? Even if security isn't necessarily critical to this application, SHA1 is in it's retirement and we shouldn't be using it for anything new. cc @sbutcher-arm

closed 2018-12-28 by brianesquilona