ARMmbed/DAPLink #102: Need clarification on ASSERT.TXT file appearing on K64F

VeliMattiLahtela, 2016-05-10:
Is this something major that we should be concerned about or is it something trivial that does not affect behaviour of the board.

No documentation on this issue.

The contents of the file:
```
cat /media/pi/DAPLINK1/ASSERT.TXT
Assert
File: ..\..\..\source\usb\msc\usbd_msc.c
Line: 237
Source: Application

```
Source application, does that tell that the application has hard faulted or something?

c1728p9, 2016-05-11:
Hi @VeliMattiLahtela, this assert indicates that more data than expected was buffered on a write event.  In this case this was a recoverable error, so the device didn't crash or hard fault.  If you can provide steps to reproduce this problem it would greatly help me in fixing this.

Here is the line this error corresponds to:
https://github.com/mbedmicro/DAPLink/blob/34182e2cce4ca99073443ef29fbcfaab9e18caec/source/usb/msc/usbd_msc.c#L237

c1728p9, 2017-01-09:
I was able to reproduce this locally and found the root cause to be a logic error in the assert. This should be fixed by #219.

closed 2017-01-09 by c1728p9