ARMmbed/DAPLink #98: Multiple K64F flashing at the same time results on non functioning device

VeliMattiLahtela, 2016-05-10:
[Jira bug](http://jira.arm.com/browse/IOTDEV-922)

This might be linked to [#94](https://github.com/mbedmicro/DAPLink/issues/94)

c1728p9, 2016-05-11:
In the issue #94 only mass storage stops working.  The serial port and HID interface still work.  Is this what you are seeing, or is the whole device locked up?  If there is a way I can reliably reproduce this then I can debug it on my end.  Alternatively, if you could do a ram dump whenever you find a device in this state it would be extremely helpful:
https://github.com/ARMmbed/post-mortem-debugging/blob/master/dump_firmware.py

VeliMattiLahtela, 2016-05-12:
Usually when this happens, the device is catatonic, no response to anything. Unplugging it and plugging it again usually solves the problem.

VeliMattiLahtela, 2016-05-18:
I couldn't get the dump_firmware working at all, as on the testing beds we do not have yotta installed.

If I try to skip the yotta parts it crashes on:

```
Traceback (most recent call last):
  File "dump_firmware.py", line 32, in <module>
    with MbedBoard.chooseBoard(frequency=10000000) as board:
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/board/mbed_board.py", line 189, in chooseBoard
    target_override, frequency)
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/board/mbed_board.py", line 161, in getAllConnectedBoards
    connected_daps = dap_class.get_connected_devices()
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/pyDAPAccess/dap_access_usb.py", line 347, in get_connected_devices
    all_interfaces = _get_interfaces()
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/pyDAPAccess/dap_access_usb.py", line 40, in _get_interfaces
    return INTERFACE[usb_backend].getAllConnectedInterface()
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/pyDAPAccess/interface/pyusb_backend.py", line 88, in getAllConnectedInterface
    product = board.product
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 841, in product
    self._product = util.get_string(self, self.iProduct)
  File "/usr/local/lib/python2.7/dist-packages/usb/util.py", line 314, in get_string
    raise ValueError("The device has no langid")
ValueError: The device has no langid
```

If you are just interested of the ram, is there any way to pull that out with pyOCD i.e. read8 command.
The only problem is that I do not know the memory map, so starting and ending points are a shot in the dark.

c1728p9, 2016-05-18:
Hi @VeliMattiLahtela, can you try the script from a laptop or another computer?  I think the device in a bad state is interfering with pyOCD.

VeliMattiLahtela, 2016-08-15:
This problem has not been seen in a while, closing the issue

closed 2016-08-15 by VeliMattiLahtela

VeliMattiLahtela, 2016-08-15:
P.S. This issue still remains, but is not an issue for us anymore.