ARMmbed/DAPLink #99: Newest firmware for K64F does not function correctly with binaries with lots of debug

VeliMattiLahtela, 2016-05-10:
[Jira bug](http://jira.arm.com/browse/IOTDEV-921)

rev 0201 firmware works perfectly with the same binary, rev 0241 does not work at all.

c1728p9, 2016-05-11:
Hi @VeliMattiLahtela, I can't reproduce this problem.  I am able to program the binary you mention and then read back the contents with a debugger and they are identical.  Is DAPLink showing any kind of an error message after programming?

c1728p9, 2016-05-11:
I did some more digging, and in that image semihosting is used.  Version 0201 partially implemented semihosting which is why the application worked with that version.  Semihosting was never official in CMSIS-DAP/DAPLink and in version 0202 and above this code was turned off, which is why you are seeing problems (this same problem occurs in all versions other than 0201).

If you use a debugger which has semihosting support then your application will run correctly with 0241.  You will also be able to see the uVisor semihosting debug statements.  One debugger you can use for this is pyOCD.

VeliMattiLahtela, 2016-05-12:
Hmmm... thats quite troublesome, as the interface that we would use to test the binary is the serial interface, and the user can't get the prompt.

SeppoTakalo, 2016-05-12:
By `lots of debug` do you mean lots of serial prints? As in debug messages?
Or are you referring to actual debugging functionality?

c1728p9, 2016-05-12:
Hi @SeppoTakalo, I think @VeliMattiLahtela, means that the image doesn't run if it is compiled for debugging.

@VeliMattiLahtela, I think it is uVisor which is using semihosting and causing this problem.  As a test I would recommend rebuilding with either uVisor turned off, or uVisor semihosting turned off.  You aren't using semihosting to debug a secure box are you?  Just the serial port right?

VeliMattiLahtela, 2016-05-13:
Actually I do not know, @SeppoTakalo might have the answer. I'm just the messenger here ;)

teetak01, 2016-05-13:
Is there some easy flag to disable the uVisor semihosting?

c1728p9, 2016-05-18:
@meriac, is there an easy way to turn off semihosting for uVisor?

meriac, 2016-05-18:
Hi Russ,

Yes, just make sure to use the release build instead of the debug build of our libraries.

Currently depending on debug/release different uVisor builds are included. You could quickly versify be copying the release.a files over the debug.a files.

Milosch

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

c1728p9, 2016-06-21:
Hi @VeliMattiLahtela, have you tried copying the release version of uvisor?  If this is cause by semihosting then I would like to close this issue.

meriac, 2016-06-21:
@c1728p9  @VeliMattiLahtela Since months uVisor has no semihosting any more, unless it's actively enabled by a uVisor app. That is consistent for the mbedmicro/mbed version of uVisor and for the yottaized-mbed-os release.

c1728p9, 2016-06-21:
I think this is a problem because it is a project from before morpheus.  Below is the output I get when I run the binary using a debugger that supports semihosting.

Serial Port:
```

   _   ___ __  __               _            _  ___  ___
  /_\ | _ \  \/  |  ___   _ __ | |__  ___ __| |/ _ \/ __|
 / _ \|   / |\/| | |___| | '  \| '_ \/ -_) _` | (_) \__
/_/ \_\_|_\_|  |_|       |_|_|_|_.__/\___\__,_|\___/|___/

[DBG ][cApp]: cmd tasklet init
/>WARNING: callback 0 took 35ms
[DBG ][core]: NS Root task Init
[DBG ][sck ]: Socket Tasklet Generated
[DBG ][mle ]: MLE Tasklet Generated
[WARN][cApp]: socket_event_handler-ARM_LIB_TASKLET_INIT_EVENT
[WARN][Asck]: socket_event_handler-ARM_LIB_TASKLET_INIT_EVENT
[WARN][cApp]: network_event_handler-ARM_LIB_TASKLET_INIT_EVENT
[WARN][cApp]: cmdline_event-ARM_LIB_TASKLET_INIT_EVENT
/>WARNING: callback 0 took 47ms
```

Semihosting:
```
uVisor mode: 0
uvisor_ram : @0x1FFF0200 (8192 bytes) [config]
             (0x1FFF0200 (8192 bytes) [linker]
```

meriac, 2016-06-21:
@c1728p9 See [vmpu.c#L83](https://github.com/ARMmbed/uvisor/blob/bd8f93f43a4d7030fd2d577b842a6b3663cc4e82/core/system/src/mpu/vmpu.c#L83): it's in vmpu_sanity_checks(), which is in vmpu_init_pre() . All of that code can't be ever reached because of [main.c#L65](https://github.com/ARMmbed/uvisor/blob/f07f7cee01bc2dc71c6eb406fef84352cc9ff250/core/system/src/main.c#L65).

I therefore infer an outdated version of uVisor is used here.

@AlessandroA

AlessandroA, 2016-06-22:
Yes, this was fixed in ARMmbed/uvisor@6b5990d (Apr 1).

c1728p9, 2016-06-22:
So can this issue be closed?
CC: @meriac, @AlessandroA, @VeliMattiLahtela

VeliMattiLahtela, 2016-06-23:
From my point, yes. @SeppoTakalo ,client teams need to use fixed uvisor.

closed 2016-06-23 by c1728p9