ARMmbed/DAPLink #105: usb power off/on produces a non attachable K64F

VeliMattiLahtela, 2016-05-19:
After a few tweaks to automation, we start too see this kind of issues is dmesg:

```
[ 1368.413390] usb 1-1.4.1: USB disconnect, device number 78
[ 1368.448472] FAT-fs (sda): unable to read boot sector to mark fs as dirty
[ 1374.788626] usb 1-1.4.1: new full-speed USB device number 80 using dwc_otg
[ 1374.868620] usb 1-1.4.1: device descriptor read/64, error -71
[ 1375.058653] usb 1-1.4.1: device descriptor read/64, error -71
[ 1375.248675] usb 1-1.4.1: new full-speed USB device number 81 using dwc_otg
[ 1375.328676] usb 1-1.4.1: device descriptor read/64, error -71
[ 1375.518685] usb 1-1.4.1: device descriptor read/64, error -71
[ 1375.708689] usb 1-1.4.1: new full-speed USB device number 82 using dwc_otg
[ 1376.128706] usb 1-1.4.1: device not accepting address 82, error -71
[ 1376.208708] usb 1-1.4.1: new full-speed USB device number 83 using dwc_otg
[ 1376.628801] usb 1-1.4.1: device not accepting address 83, error -71
[ 1376.629010] usb 1-1.4-port1: unable to enumerate USB device
```

That is after the power is turned on for the device, is DAPLINK resetting itself or what might be the reason for this kind of behaviour?

After the unable to enumerate USB device the system will not recognize the device unless the system is rebooted.

Is there any way to debug this situation i.e. wireshark from the USB interface? Is there anybody who could interpret the capture?

c1728p9, 2016-05-19:
Hi @VeliMattiLahtela, I have used wireshark to capture on windows.  I would imagine you could do the same on linux although I haven't done it before though.  If you are able to capture logs of this I would  like to take a look.

c1728p9, 2016-06-21:
Hi @VeliMattiLahtela, did you determine that this was a problem with USB hubs, or does this still appear to be a DAPLink issue?

VeliMattiLahtela, 2016-07-05:
After running multiple tests on just power cycling the device, this problem does not popup. When the device is running a client application that has a command line interface and it is accessed before power cycling this starts to appear. A workaround with RPi is to reset the data channel from RPi side.

c1728p9, 2017-09-25:
Closing this issue since I don't have enough information to reproduce or analyze the failure. Please re-open if you have any new info.

closed 2017-09-25 by c1728p9