ARMmbed/DAPLink #115: Serial issue

bridadan, 2016-06-21:
This issue is revealed in the following PR: https://github.com/ARMmbed/mbed-os/pull/382

The test case is the "echo" test that was removed. The behavior is DAPLink stops sending/receiving serial data correctly. After attempting to send and receive more serial data, eventually DAPLink returned to a good state. DAPLink also properly reset when the whole platform was power cycled.

Please contact me for me details on how to reproduce.

jupe, 2016-06-22:
are you sure that problem is not related to HW, PC(/OS) or USB hub ? We are using very heavily serial port and we haven't seen data drops anymore after daplink version 0241 I think.

mjs-arm, 2016-06-22:
Please provide more context around the daplink failure.  Does this issue manifest reliably on every execution of the test, if not, how frequently does it occur? Does this issue manifest in any other test (even intermitently?)

c1728p9, 2016-06-22:
I can confirm that this is an issue, and have a fix for it in #97.

bridadan, 2016-06-22:
@mjs-arm This failure occurs reliably when running this test, though it doesn't fail at the same point in every run of the test. The test just echos a bunch of key-value pairs that the host sends. It has three test cases within the binary, one where it does this 16 times, one where it does this 32 times, and one where it does this 64 times. The DAPLink failure occurs at some point during one of these three test cases.

PrzemekWirkus, 2016-06-23:
The was echo best behaved showed and issue in DAPLink. I've merged https://github.com/ARMmbed/htrun/pull/104 and added https://github.com/ARMmbed/htrun/pull/105 to address issues related to string tests on `htrun` side. This is a multidimensional problem, I can see it will be fixed from few angles :) `htrun`, thanks to Brian's quick investigation is almost done! :)

TeroJaasko, 2016-06-23:
I can confirm from my setup that PR #97 seemed to fix at least the occasional complete device freeze that happened after a bunch of serial IO and the debugger connection remains responsive.

closed 2016-07-11 by c1728p9