ARMmbed/DAPLink #116: Symbol missing

jamesadevine, 2016-06-30:
Hi All,

Just trying to get my dev environment set up to contribute to the firmware and i've run into an issue i can't solve without commenting assembler lines out :wink:

```
..\..\..\source\hic_hal\freescale\kl26z\armcc\startup_MKL26Z4.s(235): error: A1185E: Symbol missing
```
Which appears to be this line [here](https://github.com/mbedmicro/DAPLink/blob/master/source/hic_hal/freescale/kl26z/armcc/startup_MKL26Z4.s#L235).

This should be defined by the uvision IDE, if I am correct... I've verified it is set!

Any ideas?

c1728p9, 2016-06-30:
There is an issue like this with the latest version of progen.  If you install requirements.txt it will update progen to a supported version if it isn't already.

jamesadevine, 2016-07-01:
@c1728p9 what version of progen do you use?

c1728p9, 2016-07-01:
project-generator (0.8.15)

If I use the latest I get errors like this:
```
..\..\..\source\hic_hal\freescale\k20dx\armcc\startup_MK20D5.s(463): error: A1185E: Symbol missing
```

jamesadevine, 2016-07-02:
@c1728p9 should this be raised with the pro-gen dudes?

c1728p9, 2016-07-03:
I opened an issue earlier here:
https://github.com/project-generator/project_generator/issues/385

Looks like DAPLink will need to be updated though to fix this.

0xc0170, 2016-07-04:
> Looks like DAPLink will need to be updated though to fix this.

+1

closed 2016-07-11 by c1728p9