ARMmbed/DAPLink #118: Releases on github

Josephc3, 2016-07-02:
Would i be possible for a version of the current builds/releases to be included on the Github repo?, perhaps as a sub folder or as an external hosted link?

I have the issue that i do not have a full version of the keil MDK so cannot build the required interface firmware but it would be useful to be able to access them all up to date and in one place 

thanks,

Joe

jupe, 2016-07-02:
Release versions is there already as binary, is it enough for you:
https://github.com/mbedmicro/DAPLink/releases/tag/v0241

Josephc3, 2016-07-05:
Wow - learnt something new about github, did not know the "releases" tab existed, this is perfect! 
Exactly what i was looking for!
thanks!

closed 2016-07-05 by Josephc3