ARMmbed/DAPLink #119: USB HID

jamesadevine, 2016-07-04:
Hi all,

I'm working on some HID interface for the firmware, however there is already a default HID interface overriding the weak symbols defined elsewhere: 

https://github.com/mbedmicro/DAPLink/blob/8b2e1efe883940e3f0432fcbf41640a691881f65/source/daplink/cmsis-dap/usbd_user_hid.c

Is there a way to override this implementation without deleting/commenting the code?

c1728p9, 2016-07-05:
This is needed for CMSIS-DAP debugging.  Can you share more details on what you are wanting to do?

jamesadevine, 2016-07-05:
@c1728p9 I'm working on trying to expose a dynamic HID interface for micro:bit that allows the micro:bit to taken on a few select profiles... E.g. a keyboard and a mouse. This will be achieved through a specific mode that can be configured by the target, wherein all serial output is transformed into USB HID packets.

It should be super cool! We should have a chat about the best way to integrate the work for @alexk2k12, and myself into this repository, and at the same time talk about potential limitations of the solutions we are proposing!

c1728p9, 2016-07-13:
Closing this issue.  Feel free to email me directly to discuss this.

closed 2016-07-13 by c1728p9

carlosperate, 2017-02-03:
I was wondering if this ended up going somewhere? keyboard/mouse HID through the DAPLink would be quite cool.

c1728p9, 2017-02-03:
I haven't done any work towards it. I'm not sure if @jamesadevine has.

jamesadevine, 2017-02-19:
@carlosperate @c1728p9 I have not either, when I was working on DAPLink we were very much on the brink of running out of RAM.

DavidWhaleMEF, 2017-04-29:
We need a modular build process for DAPLink, so that people could select the modules they want for the project they are building, and flash that onto the interface chip. Not all projects need all features at the same time.

DavidWhaleMEF, 2017-04-29:
For the micro:bit, most of the target user base don't need the CMSIS-DAP interface for debugging, a better use of the resources for that demographic would be HID keyboard and HID mouse, it would enable a huge number of possible projects.

Other platforms support HID keyboard and HID mouse. I just helped a friend build a project (Minecraft sword) that had to click a mouse button. We wanted to do it with the micro:bit so it was driverless install, but not possible. We did it with an Arduino in the end as it was so simple to do. Micro:bit needs to be that simple too. We're really missing out here.