ARMmbed/DAPLink #131: NRF51_DK firmware in release package invalid

maclobdell, 2016-07-28:
For the NRF51 DK board that I have, the firmware from the release package does not work (Daplink drive does not enumerate).
0241_sam3u2c_nrf51dk_0x5000.bin

However, if the following other binary works (also from 0241 release, but not in the separate binary release package)
sam3u2c_nrf51dk_if_crc_legacy_0x8000.bin

c1728p9, 2016-07-28:
@gamnes, can you confirm that the nrf51dk expects the application to be loaded at 0x8000?

gamnes, 2016-08-02:
Depending on which bootloader you are running; the nrf51dk expects the application to be loaded at 0x8000 for the new DAPLink bootloader (MAINTANANCE), or if running with the old CMSIS-DAP bootloader (BOOTLOADER) it expects the application at 0x5000.

c1728p9, 2016-08-02:
Hi @gamnes, are new boards being created with the DAPLink bootloader?

gamnes, 2016-08-09:
New boards are created with the (old) CMSIS DAP bootloader, not the DAPLink bootloader. This is to keep compatibility with the other firmware that can be used on our boards.

maclobdell, 2016-08-09:
okay, so this issue is because the board I had was upgraded to the daplink bootloader, but most people will have the cmsis-dap bootloader out of the box.  Maybe it would still be good to have both binaries in the release package to cover both cases. @c1728p9 would you agree?

c1728p9, 2016-08-09:
There isn't a user facing way to get or update the DAPLink bootloader.  To avoid confusion the release package contains only interface firmware for existing bootloaders. I would prefer to keep it this way until new bootloaders are publicly released.

closed 2016-08-11 by maclobdell