ARMmbed/DAPLink #134: K64F with 0241 ASSERTS on usbc_msc.c

teetak01, 2016-08-08:
I am getting this kind of assert from 0241 K64F

```
Assert
File: ..\..\..\source\usb\msc\usbd_msc.c
Line: 1063
Source: Application
```

https://github.com/mbedmicro/DAPLink/blob/34182e2cce4ca99073443ef29fbcfaab9e18caec/source/usb/msc/usbd_msc.c#L1063

After this the device stops communication until it receives RESET.

This is relatively random event. I have observed it a few times during the last week. I using serial communication via Virtual Box from Win7 host and Ubuntu 16.04 client.

teetak01, 2016-08-08:
maybe related to this issue: https://github.com/mbedmicro/DAPLink/issues/102

c1728p9, 2016-08-08:
Hi @teetak01, thanks for reporting this. If you could attach logs of this occurring  it would be very helpful. Wireshark + USBpcap should allow you to capture this. Also, what do you mean by RESET? Is this a re-enumeration, a msd endpoint reset comment, a Virtual Box reset or something else?

You mention that the device stops communicating. This is only the mass storage drive right? You should still be able to debug over USB with pyOCD and you should be able to see serial data over the serial port. Can you confirm this?

Finally, Virtual Box has some known USB issues. I haven't seen any with mass storage yet, but they are fairly common with the CMSIS-DAP protocol, as Virtual Box will sometimes drop HID packets. A dropped SCSI packet might explain this assert.

jupe, 2016-08-08:
could this be related to https://github.com/mbedmicro/CMSIS-DAP/issues/93?

c1728p9, 2016-08-08:
I have only seen dropped packets when using virtual box, so if they aren't using virtual box then https://github.com/mbedmicro/CMSIS-DAP/issues/93 is probably unrelated.

jupe, 2016-08-08:
I think teemu did use raspberry with debian,,

teetak01, 2016-08-09:
No, this was on virtual box. @jupe is thinking about another possibly related issue we have been having on raspberry environment. You can seen some logs from that issue here: http://jira.arm.com/browse/IOTSYST-1146 . (K64F gets stuck in infinite? loop resending number of traces).

@c1728p9 I have to see if I can somehow reproduce this more reliably to get some logs. So far I have seen it a few times, but it may well be also something Virtual Box is causing. But also I do not remember seeing this issue before on Virtualized Ubuntu 14.04, so may also be some Ubuntu-Virtual Box issue.

The latest fail occured while I was running a testcase on K64F with CLI-testing suite. The device was in middle of sending significant amount of traces via serial port and suddenly CLI stopped receiving any data from serial port and the mass storage drive popped open as the ASSERT.TXT file appeared there. The CLI automatically sends a reset to the device when it proceeded to the next testcase and the testcases continued to execute normally from that.

teetak01, 2016-08-09:
I was now getting that fail few times in a row while re-flashing K64F. After using another K64F for a while, the first one also started to work a bit better (for now). Maybe I have a faulty hardware?

c1728p9, 2016-08-09:
Hi @teetak01, I think it is unlikely to be faulty hardware. The best way to get to the root of this issue would be to capture some USB logs to determine if the assert is from dropped USB packets.

c1728p9, 2017-01-09:
I was able to reproduce this locally and found this assert is incorrectly triggering if a valid mass storage command arrives at the right time. This should be fixed by #219 which updates the assert criteria.

closed 2017-01-09 by c1728p9

teetak01, 2017-01-09:
Ah, good to know, forgot about this completely. Thanks @c1728p9