ARMmbed/DAPLink #138: build_filesystem() called multiple times.

jamesadevine, 2016-08-22:
Hi,

`build_filesystem` is called from multiple locations, most notably on a `VFS_MNGR_STATE_CONNECTED` state.

`build_filesystem` is also called when the state is set to `VFS_MNGR_STATE_CONNECTED`, this causes the filesystem to be "built" [multiple times](https://github.com/mbedmicro/DAPLink/blob/master/source/daplink/drag-n-drop/vfs_manager.c#L177) during initialisation.

c1728p9, 2016-08-22:
Is this causing problems? As long as the filesystem isn't mounted to a PC then it should be fine to call build_filesystem multiple times. Are you working with code that has customized vfs_user_build_filesystem?

jamesadevine, 2016-08-22:
@c1728p9 My work involves communicating with the target at various points, and is built on top of @alexk2k12's work which allows custom virtual file system initialisation.

The target is reset during my custom filesystem initialisation, which means it will be reset multiple times, once on power up and once on mount, as i only expected the file system to be built once.

I of course can solve this in my implementation, but my question would be, why would you want this called multiple times?

c1728p9, 2016-08-22:
The dual initialization likely isn't necessary. Feel free to remove it. If you submit a PR for just that I'll run testing to make sure there are no regressions.

You mention that one of the calls comes from VFS_MNGR_STATE_CONNECTED. Where does the other call to build_filesystem come from?

jamesadevine, 2016-08-22:
Thanks @c1728p9, that sounds good. It will likely be in my bulk of a PR when i formulate it properly :smile: 

`vfs_mngr_init` calls `build_filesystem` which sets the state to `VFS_MNGR_STATE_CONNECTED` which does it again.

c1728p9, 2016-08-22:
Ah I see. I would recommend submitting this as a separate PR so it can be tested and verified on its own.

c1728p9, 2016-11-22:
Hi @jamesadevine, are you still planning on making this change?

jamesadevine, 2016-11-22:
@c1728p9 Unlikely, i'm out of time, and I determined there was no specific need for this to be included in mainstream DAPLink as my requirements were bespoke.

Feel free to close off this issue if you concur that a change is not required.

c1728p9, 2016-11-22:
Closing for now. Please re-open if this becomes relevant again.

closed 2016-11-22 by c1728p9