ARMmbed/DAPLink #146: Give host a chance for a graceful MSD unmount after drag-and-drop flashing

RomanSaveljev, 2016-09-02:
Once drag-and-drop flashing finishes the state of MSD endpoint seems to be toggled, which causes various shake-ups in the host side. In desktop environment it typically triggers notifications that the device has not been ejected properly before removing. 

With Linux kernels we sometimes observe that due to racing there may be several device nodes for the same MSD. This confuses tools, because one of these device nodes is a stall one and will disappear very soon. With multiple targets connected to our relatively weak raspberry pi boards the racing may really become an issue.

I think, if we could synchronize the host state before letting daplink trigger its state changes, it would help us to help more reliable system. What is your opinion?

c1728p9, 2017-09-25:
Hi @RomanSaveljev, is this still an issue? I think there were some automounter fixes which prevented the duplicate device nodes.

RomanSaveljev, 2017-09-25:
Probably, not. At least not in scope at the moment.

closed 2017-09-25 by RomanSaveljev