ARMmbed/DAPLink #162: "problem copying" on windows 8 and some Windows 10

wjzhang, 2016-09-27:
the DAPLink VFS will get the ERROR_ERROR_DURING_TRANSFER  or ERROR_TRANSFER_TIMEOUT error when "problem copying" message on Windows 8 and some Windows 10

wjzhang, 2016-09-27:
[image]

c1728p9, 2016-09-29:
Hi @wjzhang, thanks for the report. Can you provide more details? It useful to know the board you are using and the version of DAPLink (attach details.txt).

wjzhang, 2016-10-08:
failed information: "An error occurred during the transfer".
caused by one of our engineer build in dropbox. it add some attributes, when drag&drop in MSC, the Windows found it did not match. popup a dialog to continue. if press "yes", it should be working as CMSIS-DAP project. 

[image]

[DETAILS.TXT](https://github.com/mbedmicro/DAPLink/files/517328/DETAILS.TXT)

[FAIL.TXT](https://github.com/mbedmicro/DAPLink/files/517327/FAIL.TXT)

c1728p9, 2016-10-08:
It looks like this is firmware for a board that is not yet supported on DAPLink. It has the id of an EFM32 Gecko board - '2000'. Is that correct? Also, could you share the location this code came from? Finally, does this problem only occur on your board or can you get this to occur on other boards?

wjzhang, 2016-10-08:
the test Hex file and firmware are used by our board support STM32F103 MCU. it's base on lpc11u35. compatible wtih https://developer.mbed.org/teams/mbed/wiki/SWDAP. 
this issue will happened on any boards if the drag&drop file add some special attributes when some software create it. like dropbox software. 
for create this issue, you just need create your own the drag&drop hex file in  dropbox sync folder.

c1728p9, 2016-10-08:
Hi @wjzhang, I have been able to reproduce this issue. I'll see if I can track down the cause.

c1728p9, 2016-10-09:
Hi @wjzhang, I was able to identify the issue and created https://github.com/mbedmicro/DAPLink/pull/165 to fix it. Please try this out and let me know if it fixes your problem.

wjzhang, 2016-10-09:
Hi @c1728p9, thanks a lot. it fixed. BTW,  do this change effect Mac OS?

c1728p9, 2016-10-09:
Hi @wjzhang. good to hear this is fixed. I have not yet tried this on Mac OS.

c1728p9, 2016-11-22:
Closing this since #165 has been merged

closed 2016-11-22 by c1728p9