ARMmbed/DAPLink #170: Provide information on the "progen" command

DJMcMayhem, 2016-10-17:
I'm reading through the [Developers-guide](https://github.com/mbedmicro/DAPLink/blob/master/docs/DEVELOPERS-GUIDE.md) and one of the steps asks me to run 

    $ progen generate -t uvision

This fails for me, presumably because I do not have "progen" installed. While researching, I came across [this python library](https://github.com/seanballais/Progen), which seems really similar. However, after installing it, the command still fails because the "uvision" generator does not exist.

Is this the correct library? Could you provide some more details on how exactly to install/use the progen command?

c1728p9, 2016-10-17:
Hi @DJMcMayhem, it doesn't look like that is the right progen. If you install requirements.txt then the correct progen will be installed. Something like ```pip install -r requirements.txt```.

DJMcMayhem, 2016-10-19:
OK. Well, it seems like progen is being installed, but the command still fails. Here is my output from `python2 -m pip install -r requirements.txt`:

    (venv) C:\Users\Me\project\DAPLink>python2 -m pip install -r requirements.txt
    Requirement already satisfied (use --upgrade to upgrade): project-generator<0.10.0,>=0.9.8 in c:\python27\lib\site-packages (from -r requirements.txt (line 1))
    ...
    ...

However `progen generate -t uvision` gives

    'progen' is not recognized as an internal or external command,
    operable program or batch file.

As I was looking around, I did find a folder in `c:\python27\lib\site-packages` named `project_generator`, which seems to be what I'm looking for. But trying to run this gives:

    (venv) C:\Users\Me\project\DAPLink\tools>python2 -m project_generator generate -t uvision
    C:\Python27\python2.exe: No module named project_generator.__main__; 'project_generator' is a package and cannot be directly executed

    (venv) C:\Users\Me\MyProject\DAPLink\tools>

Am I even looking in the right place?

c1728p9, 2016-10-19:
It might be that the command python2 does not work properly in a virtualenv. If you run ```pip list``` (without the python2) does progen show up as installed?

c1728p9, 2016-10-24:
@DJMcMayhem were you able to get this working?

c1728p9, 2016-11-18:
Closing this issue. Please re-open if you are still having problems.

closed 2016-11-18 by c1728p9