ARMmbed/DAPLink #171: Source of flash_blob

markus-becker-tridonic-com, 2016-10-19:
I am trying to get custom STM32F401/F091 platforms supported by DAPLink on SWDAP/dipdap.

I assume I could start from lpc11u35_arm_watch_stm32f411_if and modify it to our needs. Would the flash_blob from https://github.com/mbedmicro/DAPLink/blob/master/source/target/st/stm32f411/flash_blob.c work also for F401/F091? If not, how could one get adequate blobs for ST? They do not seem to come from https://github.com/mbedmicro/FlashAlgo.

@MarceloSalazar

MarceloSalazar, 2016-10-21:
@c1728p9 could you have a look please?

c1728p9, 2016-10-24:
Hi @markus-becker-tridonic-com, the stm32f411 likely will not work with the F401 or F091. To get support for these devices they will need to be added to the FlashAlgo repo you mentioned.

markus-becker-tridonic-com, 2016-10-25:
OK, moving the issue to https://github.com/mbedmicro/FlashAlgo/issues/25

closed 2016-10-25 by markus-becker-tridonic-com

GenericWorkName, 2018-04-12:
@markus-becker-tridonic-com Hey, I'm trying to compile a firmware for stm32f401 myself on the lpc11u35.
If you managed to compile one, I'd be really happy if you could send it to me.