ARMmbed/DAPLink #176: RedBearLab BLENano fail.txt TIMEOUT

rieram, 2016-10-31:
When I plug in the MK20 programmer (via USB) to my windows 10 laptop for the RedBearLab BLENano keeps mounting and dismounting. I can load the bootloader blenano_mk20dx_interface_20140912.bin provided by the https://developer.mbed.org/platforms/RedBearLab-BLE-Nano/, but when I try to load the nrf51822_blinky_nrf51822_.hex from here as well, the mount doesn't last long enough to complete the transfer. It was mentioned on the mbed forum that there might be a bug for the loader on windows 10. Has this issue been identified and corrected? Additionally, the fail.txt file say "TIMEOUT" is displayed on the drive.
[
Please see some additional details here:](https://github.com/mbedmicro/CMSIS-DAP/issues/95)

c1728p9, 2016-11-18:
Hi @rieram, this problem has been fixed with DAPLink so there should not be problems with windows 10. Unfortunately, due to the larger size of DAPLink, the latest firmware will not fit on RBL nano boards built with the 64KB K20.

Cheong2K, 2016-12-08:
@rieram, your MK20 USB interface board is V1.5 or V1.0? If it is V1.0, you can update it to DAPLink firmware.

ayavilevich, 2017-07-15:
Hi, I have a v1.5 MK20 programmer with a K20 chip having the marking M20AFV. Is there a newer firmware available for me? Thanks.

ckunta, 2017-08-14:
I wonder what solution people has used for this problem? It appears the problem is specific to the combination of Windows 10 and mBed and MK1.5 V 1.5. I can think of several other possibilities, but want to know which approach people have taken.
- is there MK1.5 with bigger than 64KB?
- use Virtualbox with platform lower than Windows 10? (maybe Windows 7?)
- how about using Linux?
- use Arduino IDE as development, instead of mBed?

Any advice?

cederom, 2017-08-14:
Hey @ckunta :-)
* Have you tried new firmware? (no info on MKV1.5>64KB sorry).
* Are you sure this is not a VirtualBox USB issue?
* Have you tried on real Win10 laptop?
* Have you tried on Linux?
* Wasn't this issue fixed with some Windows Update patch? Win10 was known to have USB timeout issues..

ckunta, 2017-08-14:
@cederom 
- I tried with latest firmware: 0243_k20dx_rbl_0x500.bin.  The BLE Nano didn't show up after I plug that back. (May not work with DAPLink). Only when I reinstall with older CMSIS_DAP_MK20_BLENano.bin that it reconnects as MBED drive.

- I haven't done VirtualBox yet, just trying to think of that as a solution.

- Yup, tried with Windows 10 desktop and laptop. With the desktop it worked at some point 2 days ago after I installed/reinstalled those multiple .bin files. Not sure which one 'fixed' it, but when it failed again I tried to do the same thing and it didn't do anything the second time.  It never worked in the laptop. But the desktop is intermittent, works for a day, then failed again the next day. 

- Yes, the Linux seems saves the day.  My 10 years old Netbook that I converted to Linux was able to drop the .hex file without any problem and so far seems it's working. Still need to test with other code to see whether it's really working.  (Defining the question helped find the right solution for me.)

- Was there a Windows Update patch for USB?  I would like to try that. 

Thanks for the response.

cederom, 2017-08-14:
@ckunta could you please compare and verify the exact version of the windows 10 and hotfix level? You can use PowerShell: `[System.Environment]::OSVersion` and `Get-HotFix`.
Not really a Windoze fan but some hotfixes seem to introduce USB stack changes (https://support.microsoft.com/en-us/help/4022715/windows-10-update-kb4022715)..
Any idea where different behavior comes from for the same hardware/firmware would be nice :-)

ckunta, 2017-08-16:
Here is the desktop, I'll update later when I login to the laptop.

Major  Minor  Build  Revision
-----  -----  -----  --------
10     0      15063  0

Source        Description      HotFixID      InstalledBy          InstalledOn
------        -----------      --------      -----------          -----------
SOCRATES      Update           KB4022405     NT AUTHORITY\SYSTEM  6/14/2017 12:00:00 AM
SOCRATES      Security Update  KB4034662     NT AUTHORITY\SYSTEM  8/9/2017 12:00:00 AM
SOCRATES      Security Update  KB4034674     NT AUTHORITY\SYSTEM  8/10/2017 12:00:00 AM

c1728p9, 2017-09-25:
Hi @ckunta and @cederom, the issue here is a because the v1.5 BLE Nano has a K20 that does not have enough flash to fit DAPLink.  The v1.0 boards have 128KB flash while the v1.5 boards have 64k, which is not enough to run DAPLink. If you have further questions please re-open this issue.

closed 2017-09-25 by c1728p9