ARMmbed/DAPLink #179: Ring buffer overflow is not properly handed in uart.c

terhei, 2016-11-01:
When more data is written to UART ring buffer than its size is, it should not write more than it's size to USB. Buffer structutres cnt_in and cnt_out values must not be used to track how much data must be send to USB.

When buffer overflow is happening, oldest data must be overwritten with new one. When data is sent to USB, no more than buffer size must be sent. 

Used project was k20dx_frdmk64f_if and DAPlink version 0242

jupe, 2016-11-02:
I've seen this issue also. This shouldn't happen normally if host application read serial data fast enough, but if host doesn't read for a while and meanwhile device send buffer amount of data to daplink this happens. So this is kind of corner case depend on host side implementation. Should be quite easy to fix.

c1728p9, 2016-12-24:
Created PR #179 to improve overflow behavior.

c1728p9, 2017-01-02:
PR merged so this issue should be fixed.

closed 2017-01-02 by c1728p9