ARMmbed/DAPLink #180: DAPLink doesn't work with IAR EWARM 7.7?

xiongyu0523, 2016-11-02:
Hi, 

I flashed my FRDM-K64F board's K20 with '0242_k20dx_frdmk64f_0x5000.bin' image download from latest releasee. And installed mbedWinSerial_16466.exe driver. I can see 'mbed composite usb device' 'mbed Serial Port' and 'DAPLin' device in device manager. But when I select CMSIS-DAP as debugger in IAR, it shows error when launching debug session 

[image]

c1728p9, 2016-11-02:
Hi @xiongyu0523, what error is shown? Also, does this start working again if you revert to an older version?

c1728p9, 2016-11-18:
Hi @xiongyu0523, any updates on this?

c1728p9, 2017-03-09:
@xiongyu0523 I can't reproduce this problem. Please re-test and let me know if you are still experiencing this issue. If you are please re-open this issue. There have been many exporter fixes in mbed-os so this is likely fixed.

closed 2017-03-09 by c1728p9