ARMmbed/DAPLink #182: Flash verification during programming

c1728p9, 2016-11-09:
DAPLink should verify each block flashed by reading back the flash. This should either be turned on by default or enabled by a configuration value.

jupe, 2016-12-05:
IOTDEV-925

flit, 2016-12-05:
Note that some flash algos do this themselves. For instance, this is the Kinetis flash algo page program functions:
```c
uint32_t ProgramPage(uint32_t adr, uint32_t sz, uint32_t *buf)
{
    int status = FLASH_Program(&g_flash, adr, buf, sz);
    if (status == kStatus_Success)
    {
        // Must use kFlashMargin_User, or kFlashMargin_Factory for verify program
        status = FLASH_VerifyProgram(&g_flash, adr, sz,
                              buf, kFLASH_marginValueUser,
                              NULL, NULL);
    }
    return status;
}
```

sg-, 2016-12-23:
> Note that some flash algos do this themselves. For instance, this is the Kinetis flash algo page program functions:

Agreed. The flash algo should be responsible for verify. Its a little disappointing to see it listed as optional 👎 

http://arm-software.github.io/CMSIS_5/Pack/html/flashAlgorithm.html

c1728p9, 2017-01-10:
Created PR #221 to resolve this.

c1728p9, 2017-01-10:
@mbed-bot: TEST

closed 2017-01-10 by c1728p9