ARMmbed/DAPLink #185: post_computer_crc starting address.

kgills, 2016-11-11:
This code is from tools/post_compute_crc.py.

```
    if start == 0x8000 or start == 0x88000:
        pad_addr = start - 0x3000
        # Create hex file
    elif start == 0 or start == 0x80000:
        pass
    else:
        assert 0, "Unsupported start address 0x%x" % start
```

Can someone explain what's going on here. It seems to be creating hex files for the interface firmware and not for the bootloader. The issue that I'm having is that when the interface firmware starting address is not at 0x8000 or 0x88000 I get an error.

The HIC I'm working with has a flash sector size of 0x2000. I don't have enough room for the bootloader code and have a page for the "CONFIG_ADMIN" page.

c1728p9, 2016-11-11:
Hi @kgills, the first 'if' is for creating legacy builds. Firmware for new boards shouldn't need this. The 'elif' is for versions which don't need a legacy build so all addresses are accounted for.

Addresses which cause a legacy build to be created in addition to the regular build:
0x8000 - Legacy application start for the K20DX and KL26Z
0x88000 -  Legacy application start for the atsam3u

Ignored addresses:
0x0 - Ignores Bootloader for K20DX and KL26Z, interface for lpc11u35
0x80000 - Ignores bootloader for the atsam3u.

kgills, 2016-11-14:
Would it be acceptable to remove the assertion? If only legacy builds need these hex files, and their starting addresses won't change, I don't think we need the assertion.

c1728p9, 2016-11-14:
Yeah, you can either remove the assert or add the new offset to the elif case. If the assert is left in then it will ensure all offsets are accounted for.

kgills, 2016-11-14:
I'll create a PR to remove the assertion. Since only the legacy builds need this and their starting addresses won't change, we won't have to add a new starting address every time a new HIC wants to use a different starting address.

c1728p9, 2016-11-22:
@kgills can this be closed since #186 has been merged?

closed 2016-11-22 by kgills