ARMmbed/DAPLink #200: Assert filenames missing from interface on Kinetis

c1728p9, 2016-12-03:
Assert filenames were unintentionally disabled for both the interface and bootloader in #151 . They should only be disabled for the K20 and KL26 bootloader to safe space. 

CC @flit

flit, 2016-12-03:
@c1728p9 Oops, sorry, how did I screw it up?

c1728p9, 2016-12-04:
I think it is because these files are common to both the bootloader and interface:
records/hic_hal/k20dx.yaml
records/hic_hal/kl26z.yaml

flit, 2016-12-04:
What about splitting `DAPLINK_NO_ASSERT_FILENAMES` into `DAPLINK_NO_BL_ASSERT_FILENAMES` and `DAPLINK_NO_IF_ASSERT_FILENAMES`?

flit, 2016-12-10:
@c1728p9 Any feedback on this?

c1728p9, 2016-12-10:
Sorry, missed your comment. That sounds like a good solution.

c1728p9, 2016-12-20:
I created #203 as a simple fix for this. Hopefully it will never get to the point where asserts need to be turned off on the interface.

closed 2016-12-22 by c1728p9

flit, 2016-12-22:
Ok, thanks. Sorry I didn't have a chance to get to it right away.