ARMmbed/DAPLink #201: Windows 10 host renders DAPLink useless

ErichStyger, 2016-12-10:
I'm facing a sporadic issue with DAPLink under Windows 10. Unfortunately there is no clear reproducible pattern, but it happens enough that it is really annoying. So hopefully this gives hints into identifying the issue.
I'm using DAPlink on multiple NXP boards (FRDM-K22F, FRDM-K64F and others based on the NXP K20 OpenSDA circuit). The problem is that Windows 10 seems to send packets to the DAPlink while it is in BOOTLOADER mode. These packets somehow are taken by the Bootloader as application files/data and it seems to start programming the application. But this of course does not work. As a result, the bootloader  seems to overwrite itself: the board does not show up as bootloader any more.
The only solution/workaround is to re-flash the DAPLink bootloader with a debug probe.

The issue occurs whenever the DAPLink is in BOOTLOADER mode, sometimes it takes 10-20 seconds until it happens. The issue is easier to reproducible if the DAPlink microcontroller is in constant reset as I faced the issue much more often when I bring up a new target microcontroller (which stays in reset initially).

The same issue already exists with the original NXP OpenSDA bootloader (see https://mcuoneclipse.com/2016/08/01/bricking_and_recovering_opensda_boards_in_windows_8_and_10/). So I think the issue is the same for the DAPLink bootloader.

I hope this helps,
Erich

c1728p9, 2016-12-10:
Hi @ErichStyger, thanks for reporting this. What version of the bootloader are you using? If you could attach DETAILS.TXT and MBED.HTM it would help me identify it.

ErichStyger, 2016-12-10:
Sure, I thought I had it attached in the first report, so here as plain text:

# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 0000000026294e45001a000e58a9001d3ee1000097969900
HIC ID: 97969900
Auto Reset: 0
Automation allowed: 1
Daplink Mode: Bootloader
Bootloader Version: 0241
Git SHA: be5afd34db2efe1e08d72fa23b3a68ab2625d9a3
Local Mods: 1
USB Interfaces: MSD
Bootloader CRC: 0x439c8da1
Interface CRC: 0x07911068

<!doctype html>

<html>
<head>
<meta charset="utf-8">
<title>mbed Website Shortcut</title>
</head>
<body>
<script>
window.location.replace("http://www.nxp.com/opensda");
</script>
</body>
</html>

c1728p9, 2016-12-10:
Based on the info you send, the bootloader version you are using should have the Windows 10 problems of previous versions fixed so this might be a new bug.  Can you send me a binary image of the bootloader you are using so I can try it locally? Also, if you are able to dump flash when this happens and send me a log of it that would be very helpful.

ErichStyger, 2017-01-18:
attached is the binary.

[OpenSDA 2.2 Daplink BOOTLOADER for K20DX 0x8000.zip](https://github.com/mbedmicro/DAPLink/files/713134/OpenSDA.2.2.Daplink.BOOTLOADER.for.K20DX.0x8000.zip)

c1728p9, 2017-02-16:
Hi @ErichStyger, I gave this firmware a try on Windows 10 on a FRDM-K64F but I was unable to reproduce any problems. You mentioned that this occurs more often when the DAPLink microcontroller is resetting often. I did cycle testing which resets the K20DX and let it run for 700+ cycles and didn't see any problems.

Is it possible that the malfunctioning bootloader is an older one from the CMSIS-DAP repo? I have seen a problem similar to the one you described, but it was older firmware. In that case it was the bootloader that shipped with a FRDM-K66F, which I believe was built expecting an application offset of 0x5000 not 0x8000.

ErichStyger, 2017-02-16:
Thanks for checking. That bootloader is using an offset of 0x8000, so must be a newer one.
The semester will start again next week, so I will see if there is any more occurence of the issue.

ashok-rao, 2017-05-12:
@c1728p9 , I'm seeing this as well. I bricked a FRDM-K22F just now and trying to revive it with a J-Link..

dhwalters423, 2017-09-14:
@c1728p9 I have just encountered this issue on the FRDM KW24D- device is bricked.

c1728p9, 2017-12-12:
Hello @dhwalters423 @ashok-rao and @ErichStyger, there is a blog post on mbed-os about this:
https://os.mbed.com/blog/entry/DAPLink-bootloader-update/

It goes over how to prevent bricking and how to recover boards which have been bricked.

closed 2017-12-16 by sg-