ARMmbed/DAPLink #206: What is the assumed environment for the commands shown in DEVELOPERS-GUIDE.md

OSHChip-Owner, 2016-12-23:
For the commands shown in the DEVELOPERS-GUIDE.md , what is the assumed environment.
I'm using Windows 7 Pro x64, and I am using the Git Shell command window (which is Microsoft's
PowerShell).  This was installed from https://desktop.github.com/  which installed both the
GUI interface for Github, and the above mentioned Git Shell command window.

Unlike the examples in the guide, instead of a prompt of '$' , I get prompts like this:
   D:\GitHub_Clone\DAPLink [master ≡ +2 ~1 -0 !]>

The issue arises at Step 2, where the activate and deactivate commands are quoted: 
   "venv/Scripts/deactivate"

When I enter them as above, the text is echoed back without the quotes, but no command
is actually run.  If I leave the quotes off, the commands run correctly.

Is this an error in DEVELOPERS-GUIDE.md or should I be using some other command
window (in which case maybe that should be documented in the guide)?

Thanks.

c1728p9, 2017-01-06:
I typically use cmd.exe, but other shells should work too. I'll remove the quotes so the directions work in more configurations.

c1728p9, 2017-01-09:
This is fixed by #220

closed 2017-01-09 by c1728p9