ARMmbed/DAPLink #214: Building the binary for NXP TWR-LS1021A board

skhasan, 2017-01-02:
I need the latest binary based on the latest DAPLink source for TWR-LS1021A board which is based on MK20DX128VFM5. I have followed the guidelines mentioned by DEVELOPERS-GUIDE.md. The problem is that I could not find this board specific code in the "source/board" directory. As this is based on the K20DX target, I have tried with "k20dx_bl.bin" but the board did not boot up.

Has anyone tried the DAPLink source for TWR-LS1021A board?
Do I need to port this source for the TWR-LS1021A board?

c1728p9, 2017-01-06:
Hi @skhasan, I haven't  tried DAPLink with this board, although DAPLink does support two other tower boards - ```twrkl28z72m``` and ```twrke18f```. The file k20dx_bl.bin should work on other tower boards. How are you loading it?

skhasan, 2017-01-09:
Could you please let me know how to flash the bootloader (k20dx_bl.bin)? Currently my board has CMSIS-DAP (not DAPLink ) bootloader preinstalled. That is why the DAPLink interface firmware (build 203) is not working for my board. In my understanding, I should flash the DAPLink bootloader first but I do not know how to flash it. Please help me.

c1728p9, 2017-01-09:
You'll need to program the K20 over swd with an external debugger using uVision.

c1728p9, 2017-09-25:
Closing due to inactivity. @skhasan please reopen if you have further questions.

closed 2017-09-25 by c1728p9