ARMmbed/DAPLink #223: Doesn't display "DAPLINK" message when upload DAPLink firmware using LPC11U35 ISP Boot

kaizen8501, 2017-01-13:
Hi, I have a problem using DAPLink on LPC11U35.

I try to update to DAPLink from CMSIS-DAP using WIZwiki-W7500. WIZwiki-W7500 use LPC11U35 for interface. This is schematic of WIZwiki-W7500.

[WIZwiki-W7500 V1.1 Schematic (PDF)](http://wizwiki.net/wiki/lib/exe/fetch.php?media=products:wizwiki_w7500:wizwiki_w7500_v1.1.pdf)

It is success when I upload firmware of LPC11U35 using JTAG(ULINK2). It can display "DAPLINK" well in my PC as this picture.

[image]

But If I upload binary using LPC11U35 ISP Boot( CRP DISABLD ), It doesn't display "DAPLINK" after done. It only display "CPR DISABLd" in my PC.

[image]

Is there anybody who help me?

c1728p9, 2017-01-13:
Hi @kaizen8501, what image are you loading? If you built them with uvision then you'll need to load the image that ends with "_crc.bin".

kaizen8501, 2017-01-16:
Thank you. I solved it. First my problem is that python doesn't have "intelhex" module. So I could't get "_crc.bin". So I installed "intelhex" module to python and I upload "_crc.bin" to my board. I works well. Thank you.

c1728p9, 2017-03-07:
Hi @kaizen8501, it sounds like this issue has been resolved, so I'm closing it. If this is a mistake, please re-open.

closed 2017-03-07 by c1728p9