ARMmbed/DAPLink #224: Potential null-pointer dereference when receiving serial data before USB is connected

devanlai, 2017-01-15:
While doing development on an STM32F103 HIC based on #133, I noticed that if the target MCU sends serial data very early on, the DAPLink firmware could encounter a bus fault.

I'm currently testing with this target program:
```
#include "mbed.h"

Serial pc(USBTX, USBRX); // tx, rx
int main() {
    pc.printf("Hello!\r\n");
    while(1) {
        wait(0.2);
        pc.printf("Bye!\r\n");
    }
}
```

After doing some digging, I found the following:

- When the USB stack is initialized, the CDC-ACM driver calls `USBD_CDC_ACM_Initialize()` to run the user hook and initialize the UART, but it does not fully initialize the CDC-ACM driver's internal state until a USB reset event calls `USBD_CDC_ACM_Reset()`
- In-between initializing the USB stack and connecting, any serial data received will be written to the uninitialized CDC-ACM circular buffer, triggering a bus fault when it tries to access 0x00000000.
- This happens consistently on the STM32F103 HIC because it initializes the UART with a default 9600 baud configuration even before connecting over USB, whereas other HICs enable the UART without setting the baud rate.
- On an LPC11U35 based HICs, this doesn't appear to be a problem normally. However, I can hang it if I increase the USB connection delay from eleven 90ms ticks to twenty, and applying my patch fixes that hang, so I suspect that it's still susceptible to the same issue, but it's just harder to send serial data that the partially configured UART will accept as valid.

I'm currently using this patch as a quick work-around:
```diff
 __weak int32_t USBD_CDC_ACM_Initialize(void)
 {
+    ptr_data_to_send = USBD_CDC_ACM_SendBuf;
+    ptr_data_sent = USBD_CDC_ACM_SendBuf;
     return (USBD_CDC_ACM_PortInitialize());
 }
```

Long-term, I think it might make more sense to not turn on the UART until a CDC-ACM set-line-coding request is received, since there's no telling what baud-rate the host expects anyways. Any data received before the line coding is set is potentially garbage.

c1728p9, 2017-03-24:
Thanks for the report. I was able to reproduce the crash on an LPC11u35 and confirm that initializing those variables fixes the problem. I created PR #253 for this.

Although less than idea, the early initialization of the UART should not effect users, since the existing implementations clear the UART buffer on baudrate change.

c1728p9, 2017-09-25:
Closing this issue since it has been fixed

closed 2017-09-25 by c1728p9