ARMmbed/DAPLink #225: Failed to flash FRDM-KW24D

SeppoTakalo, 2017-01-30:
I was in assumption that `FRDM-KW24D` is supported.

However, the release package contains `0243_k20dx_frdmkw24f_0x8000.bin` binary. I assumed that is it only typo, and tried to flash and got following error on Mac:
[image]

It did not update the firmware. Just rebooted the bootloader.

Then I tried again in Linux:
I did not see any error messages, but following was visible in `dmesg`
```
[10529801.975734] usb 3-5.1: new full-speed USB device number 13 using xhci_hcd
[10529801.994286] usb 3-5.1: New USB device found, idVendor=0d28, idProduct=0000
[10529801.994293] usb 3-5.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[10529801.994296] usb 3-5.1: Product: MBED BOOTLOADER
[10529801.994299] usb 3-5.1: Manufacturer: MBED
[10529801.994301] usb 3-5.1: SerialNumber: ffff1000C82F6E5FD82F91A0
[10529801.995390] usb-storage 3-5.1:1.0: USB Mass Storage device detected
[10529801.995511] scsi15 : usb-storage 3-5.1:1.0
[10529802.991977] scsi 15:0:0:0: Direct-Access     MBED     BOOTLOADER            PQ: 0 ANSI: 2
[10529802.992326] sd 15:0:0:0: Attached scsi generic sg4 type 0
[10529802.992627] sd 15:0:0:0: [sdd] 240 512-byte logical blocks: (122 kB/120 KiB)
[10529802.992844] sd 15:0:0:0: [sdd] Write Protect is off
[10529802.992848] sd 15:0:0:0: [sdd] Mode Sense: 03 00 00 00
[10529802.993076] sd 15:0:0:0: [sdd] No Caching mode page found
[10529802.993080] sd 15:0:0:0: [sdd] Assuming drive cache: write through
[10529802.994713] sd 15:0:0:0: [sdd] No Caching mode page found
[10529802.994717] sd 15:0:0:0: [sdd] Assuming drive cache: write through
[10529803.002725]  sdd:
[10529803.003801] sd 15:0:0:0: [sdd] No Caching mode page found
[10529803.003805] sd 15:0:0:0: [sdd] Assuming drive cache: write through
[10529803.003809] sd 15:0:0:0: [sdd] Attached SCSI removable disk
[10529829.404406] usb 3-5.1: USB disconnect, device number 13
[10529829.438580] FAT-fs (sdd): unable to read boot sector to mark fs as dirty
```

Then board just lid all leds on board. It did not restart the bootloader.
Board did not reappear anymore, even after unconnecting&reconnecting. Now only bootloader mode works(connect with RESET pressed).

SeppoTakalo, 2017-01-30:
@sg- @c1728p9 @maclobdell Please check.

SeppoTakalo, 2017-01-30:
Using the bootloaders provided by NXP, the result is the same.
http://www.nxp.com/products/software-and-tools/hardware-development-tools/startertrak-development-boards/opensda-serial-and-debug-adapter:OPENSDA#FRDM-KW24D512

The default firmware flashes OK, but version 0242 does not.

c1728p9, 2017-01-30:
@flit should the FRDM-KW24F firmware in DAPLink be renamed to FRDM-KW24D? Also, it looks like the offset specified in info.py is wrong for the KW24F - currently it is 0x8000, but the bootloader expects 0x5000. Can you confirm? It might also be worth checking some of the other new freescale targets to make sure they have the right offset.

flit, 2017-02-02:
@c1728p9 Yes, it should be KW24D, not F. No FPU on that one.

c1728p9, 2017-02-02:
@flit when you get a chance could you submit a PR to update this and any other NXP targets with the wrong name?

flit, 2017-02-02:
Sure thing. I don't think there are any other incorrect names.

c1728p9, 2017-09-25:
This is fixed by #228. Please re-open this issue if this is not the case.

closed 2017-09-25 by c1728p9