ARMmbed/DAPLink #229: micro:bit 0243 has html that points to https://www.microbit.co.uk/

OwenBrotherwood, 2017-02-14:
the html redirect is to https://www.microbit.co.uk/ I wonder if this is correct?

```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9900000036424e4500413014000000140000000097969901
HIC ID: 97969901
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0243
Git SHA: b403a07e3696cee1e116d44cbdd64446e056ce38
Local Mods: 0
USB Interfaces: MSD, CDC, HID
Interface CRC: 0x14256f44
Remount count: 1

```

jaustin, 2017-02-15:
@OwenBrotherwood @c1728p9 that's a good point, future builds of DAPLink should head to a microbit.org page that can deal with firmware updates based on firmware version, but that's not there yet. @c1728p9 We'll need to build a new place to hit on microbit.org for now let's use microbit.org/code?mbedcode=

OwenBrotherwood, 2017-02-15:
(Did anyone mention gcc ...)

c1728p9, 2017-03-07:
@jaustin so would it look something like this?
```
window.location.replace("https://www.microbit.org/code?mbedcode=99000243");
```

jaustin, 2017-03-07:
Yea, that’d be good thanks! We can make ‘mbedcode’ whatever we want at this stage, but we’re going to live with it for a long time, so we should be sure about it ;)

Jonny

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

closed 2018-12-28 by brianesquilona