ARMmbed/DAPLink #232: How to make DAPLink firmware support STM32F412ZGT6 and STM32L443 MCUs?

Erick-Huang, 2017-02-17:
We’re developing DAPLink firmware with the board based on NXP LPC11U35 and we also follow the design of https://developer.mbed.org/platforms/SWDAP-LPC11U35/ as reference to develop the DAPLink board. However, according to the spec of SWDAP-LPC11U35, it only supports main targets like NXP K64F/NXP LPC1768/Nordic nRF51822.

We were wondering how we can make our DAPLink board support STM32F412 and STM32L443 targets? What is required from our side to develop the DAPLink firmware?

Awais174, 2017-04-26:
Hello, 
Please go through this: https://github.com/mbedmicro/DAPLink/blob/master/docs/PORT_TARGET.md
You will have to update target_reset.c, target.c after creating flash_blob.c through https://github.com/mbedmicro/FlashAlgo.
Though, currently flashAlgo only supports STM32F4XX series, and you may have to update it for STM32LXX series. 
I hope that helps.

c1728p9, 2017-09-25:
Please re-open this issue if you have further questions @Erick-Huang

closed 2017-09-25 by c1728p9