ARMmbed/DAPLink #236: DIPDAP-LPC11U35 firmware lpc11u35_501 not built?

inindev, 2017-02-25:
Reading the project page for the DIPDAP debug probe:
https://developer.mbed.org/platforms/DIPDAP-LPC11U35

It states "This debug probe is supported by the DAPLink project."  In the upper right hand part of the page, I see "To compile a program for this board, use lpc11u35_501 as the target name."  This leads me to believe I should be looking in the quarterly "0243_release_package_b403a07e.zip" for a firmware file named something like "0243_lpc11u35_501_0x0000.bin"

Note: I dont own a copy of Keil (nor a Windows machine for that matter), so I am unable to compile DAPLink firmware on my own.  I see support for gcc has been discussed but that effort stalled: https://github.com/mbedmicro/DAPLink/issues/197

Does anyone know why the lpc11u35_501 firmware isn't being built in the quarterly release or where I need to go to find this file?  Any help is much appreciated.

inindev, 2017-02-26:
Through a bit of trial-and-error, I discovered that the `0243_lpc11u35_archpro_0x0000.bin` firmware works on the `DIPDAP-LPC11U35` debug probe.  Debugging with pyOCD/gdb seems to work just fine on my Mac.

There is an issue with drag-drop programming, however:  The target device does not reset after being programmed and the USB connection gets reset with a timeout error.  Manually resetting the target is all that is required.

c1728p9, 2017-02-28:
Hi  @inindev, DIPDAP is built from DAPLink but is not main line on master, since it requires some custom modifications (like turning off mass storage).

Most lpc firmware images will work with the DIPDAP since the hardware is very similar. The archpro firmware expects to have a LPC1768 attached, so you will only be able to program it with drag-n-drop.

c1728p9, 2017-09-25:
Please re-open if you have further questions.

closed 2017-09-25 by c1728p9