ARMmbed/DAPLink #247: kl82z targte.c: ram_end address mistake

Hoohaha, 2017-03-14:
```
  .ram_start      = 0x1FFFA000,
  .ram_end       = 0x2002000,
```
Looks like the .ram_end make a mistake?
Anyone can help to check?

c1728p9, 2017-03-14:
.ram_end should be the address after the end of ram. For example if  ```.ram_start      = 0x20000000``` and ram size is 0x8000 then you should set ```.ram_end  = 0x20008000```

In your comment above, `.ram_end` is at a lower address than `.ram_start` which is wrong. Is this a typo or did you see this in the code? I looked at the K82F's target.c, but didn't see anything wrong there.

Hoohaha, 2017-03-16:
sorry, my expression is not clear, i mean the .ram_end missing 1bit.

c1728p9, 2017-03-24:
Hi @Hoohaha, in my example earlier, if `.ram_start      = 0x20000000`and ram size is 0x8000 then you should set `.ram_end  = 0x20008000` and not `.ram_end  = 0x20007ffff`. Does that answer your question?

Hoohaha, 2017-03-27:
uh... ram_start is 8bit, but .ram_end is 7bit, this should be a bug!

c1728p9, 2017-03-27:
I understand what you mean now. This is a bug. @flit can you confirm that the correct address should be 0x20020000 [here](https://github.com/mbedmicro/DAPLink/blob/master/source/target/freescale/kl82z/target.c#L34)?

flit, 2017-03-27:
The KL82 has 96kB of SRAM total, split 24kB/72kB. So the correct address range is 0x1FFF_A000—0x20012000.

c1728p9, 2017-10-19:
I created #331 to fix this.

c1728p9, 2017-11-21:
Thanks for reporting this @Hoohaha. I'm closing this since it has been fixed.

closed 2017-11-21 by c1728p9