ARMmbed/DAPLink #248: launch_uvision.bat  not work

neomissing, 2017-03-15:
I want to use the MDK (uVision) IDE to work with the DAPLink code,
I have installed the software environment，But launch_uvision.bat  not work,
It fails to work well，
I hope to provide the same MDK project as  CMSIS DAP,
Thanks very much

c1728p9, 2017-03-15:
Hi @neomissing, what error do you get?

neomissing, 2017-03-15:
EEROR see picture1，reasult as  picture2，can your give me DAPLink mdk project？

neomissing@163.com
 
From: Russ Butler
Date: 2017-03-15 21:59
To: mbedmicro/DAPLink
CC: neomissing; Mention
Subject: Re: [mbedmicro/DAPLink] launch_uvision.bat not work (#248)
Hi @neomissing, what error do you get?

neomissing, 2017-03-15:
[

neomissing, 2017-03-15:
/d/DAPLink-master/tools/launch_uvision.bat

D:\DAPLink-master>setlocal

D:\DAPLink-master>cd D:\DAPLink-master\tools\..\

D:\DAPLink-master>call env\Scripts\activate
Doing pip install
Collecting project_generator<0.10.0,>=0.9.8 (from -r requirements.txt (line 1))
  Using cached project_generator-0.9.10.zip
Collecting mbed_ls (from -r requirements.txt (line 2))
  Using cached mbed-ls-1.2.12.zip
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\ADMINI~1\AppData\Local\Temp\pip-build-_asvno7t\mbed-ls\setup.py", line 38, in <module>
        long_description=read('README.md'),
      File "C:\Users\ADMINI~1\AppData\Local\Temp\pip-build-_asvno7t\mbed-ls\setup.py", line 33, in read
        return open(os.path.join(os.path.dirname(__file__), fname)).read()
    UnicodeDecodeError: 'gbk' codec can't decode byte 0x9d in position 14213: illegal multibyte sequence

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in C:\Users\ADMINI~1\AppData\Local\Temp\pip-build-_asvno7t\mbed-ls\

c1728p9, 2017-03-15:
Hi @neomissing it looks like there is a problem with installing required python modules. You might want to try forcing a rebuild of the virtual environment by deleting the directory 'env' and launch_uvision.bat again.

neomissing, 2017-03-16:
I tried, spent a lot of time, but still can not work, you can directly give the project

c1728p9, 2017-03-16:
Done. You'll still want to get to the root cause of the python problems. Can you try running these commands manually to what goes wrong? You might have something wrong with your python install. What version of python are you using?

```
virtualenv testenv
call testenv\Scripts\activate
pip install -r requirements.txt
```

neomissing, 2017-03-17:
Thank you very much. I have tried again, but the result is the same. Another problem is that the project is missing k20dx_frdmk22f_if_crc.c files,Project is not wrong?

neomissing@163.com
 
From: Russ Butler
Date: 2017-03-17 01:00
To: mbedmicro/DAPLink
CC: neomissing; Mention
Subject: Re: [mbedmicro/DAPLink] launch_uvision.bat not work (#248)
Done. You'll still want to get to the root cause of the python problems. Can you try running these commands manually to what goes wrong? You might have something wrong with your python install. What version of python are you using?
virtualenv testenv
call testenv\Scripts\activate
pip install -r requirements.txt

closed 2017-03-20 by neomissing

c1728p9, 2017-03-20:
Hi @neomissing, I noticed you recently closed this issue. Were you able to solve your problem?

neomissing, 2017-03-20:
Last night I tried many times to solve this problem, the problem is that Python and pip do not match, thank you for your help!!!

neomissing@163.com
 
From: Russ Butler
Date: 2017-03-20 10:21
To: mbedmicro/DAPLink
CC: neomissing; Mention
Subject: Re: [mbedmicro/DAPLink] launch_uvision.bat not work (#248)
Hi @neomissing, I noticed you recently closed this issue. Were you able to solve your problem?