ARMmbed/DAPLink #249: NXP Link based on LPC11U35 or LPC4322

tz-arm, 2017-03-23:
The link to NXP website mentioned the NXP LPC LINK and LINK2, but I found that they are based on LPC3154 and LPC4370, why we write LPC11U35 and LPC4322 here?

flit, 2017-03-23:
Because DAPLink is designed to run on boards with an on-board debug probe using LPC11U35 and LPC4322, not the LPC-LinkI/II boards. LPC-LinkI/II has its own firmware. (Though DAPLink could be ported rather easily.)

c1728p9, 2017-09-25:
Closing this issue. @tz-arm please reopen if you have more questions.

closed 2017-09-25 by c1728p9

rbarzic, 2019-06-02:
Sorry to jump into that but I really think that any mention to  the LPC-LinkII board should be removed from this site - This is very misleading  - I thought the LPC-LinkII was officially supported by the DAPLink team (which is apparently wrong) and EmbeddedArtists/NXP claims support for DAPLink without mentioning that their software is actually custom, not available on their web site  and locked to debug NXP devices only.  (something that Embedded Artists does not mentioned on their website).  Now, I need to port  DAPLink to this board which means having access to a KEIL license....:-(

0xc0170, 2019-06-03:
> Sorry to jump into that but I really think that any mention to the LPC-LinkII board should be removed from this site

@rbarzic Can you add here references here? We shall review them.

40Grit, 2019-06-03:
If you are in a bind, and can't wait for a preferable solution.
Mbed Studio Comes with ARMC6.
Daplink doesn't use Mbed OS so building it with Mbed Studio's AC6 might not be within the license.