ARMmbed/DAPLink #251: K20 and KL26 Is daplink_addr right?

neomissing, 2017-03-24:
K20 :
#define DAPLINK_RAM_START               0x1FFFE000
#define DAPLINK_RAM_SIZE                0x00004000
KL26 :
#define DAPLINK_RAM_START               0x1FFFF000
#define DAPLINK_RAM_SIZE                0x00004000
I read books, found that the two opposite。K20 and KL26 Is daplink_addr right?

c1728p9, 2017-03-24:
Hi @neomissing, according to the reference manual for these devices, the K20's start address is `0x2000_0000 – SRAM_size/2` and the KL26's start address is `0x2000_0000 – SRAM_size/4`. So with a ram size of 0x4000 then the starting address of each is as follows:
K20: `0x2000_0000 – SRAM_size/2` = 0x1FFFE000
KL26: `0x2000_0000 – SRAM_size/4` = 0x1FFFF000

So these addresses should be correct.

c1728p9, 2017-09-25:
Closing this issue. Please re-open if you still have questions.

closed 2017-09-25 by c1728p9