ARMmbed/DAPLink #260: How to change DAPLINK baud-rate at run time

akrnvm, 2017-04-24:
How DPALINK will work in different baud-rates ?

In my DAPLINK it fixed the baud-rate as 9600,hence its not able to prints different baud-rate applications messages(115200). how can I modify DAPLINK code to common for all baud-rates without changing in serial driver (for baud-rate )in each time.

DAPLINK with fixed boud-rate is expect behavior or shall i need to change my driver to handle all baud-rates dynamically ?

Please anyone can clarify

flit, 2017-04-24:
Baud rate is specified via USB CDC.

pranavshandilya, 2017-04-24:
[https://developer.mbed.org/questions/77827/Daplink-baud-rate-changing/](url)
@akrnvm The same question was actually posted to the developer site today.

I did the same thing in my DAPlink, 
In uart.c, under uart_initialize function Baud rate is fixed to 9600.

@flit Could you please elaborate this answer?

c1728p9, 2017-04-26:
Hi @akrnvm and @pranavshandilya, the baudrate in uart_initialize doesn't matter, the host does not read data until it configures the UART (which sets the baudrate). This is done in the function [uart_set_configuration](https://github.com/mbedmicro/DAPLink/blob/master/source/hic_hal/uart.h#L81).

To change the baudrate you'll need to select a new baudrate in the terminal program you are using (programs like CoolTerm, Tera Term, PuTTY, Hyperterminal, screen, ...). More info can be found in the [serial communication docs](https://docs.mbed.com/docs/mbed-os-handbook/en/latest/getting_started/serial_communication/).

pranavshandilya, 2017-04-27:
Hi, @c1728p9 Thanks

I'll try your suggestion, Once it is done I'll inform you.

akrnvm, 2017-04-28:
RTS/CTS flow-control pins are required for this? in my target board only Tx pin and Rx pin are connected with DAPlink. 

in "uart_set_configuration" function i'm re-init my UART with config->baudrate, but its not reflect in my output.
 is this "config->baudrate" value is "9600" , "115200",etc values or some thing else?

c1728p9, 2017-04-28:
Hi @akrnvm, what terminal program are you using on the PC to open the serial port?

akrnvm, 2017-05-02:
@c1728p9 thank you for your support.
i'm using Tera-Term as terminal program in my PC.
Now the issue got resolved, it was a code mistake from my side.

closed 2017-05-02 by akrnvm