ARMmbed/DAPLink #261: DAPLink on custom board(LPC1768).

pranavshandilya, 2017-04-24:
1 year ago, I have created bootloader and interface firmware for a custom controller(DAPLink V203) and it's working fine with Target controller LPC1768. I'm able to do Automated testing and everything.

But now same code ported to new DAPLink(V243) with reference of bootloader sam3u2c_bl and interface firmware sam3u2c_mkit_dk_dongle_nrf5x_if, and it's not working.

Individually it's working means when I download bootloader to controller "MAINTANCE" drive is showing but after copying interface firmware into the drive, it disappears after restart nothing will happen.
But the same interface firmware is working with an old bootloader(V203).

ROM size: 256KB
RAM size: 48KB

Why it's not working with a combination of the new bootloader(V243)?
Please anyone can help me for this, Thanks..!

daplink_addr.h

```
/**
 * @file    daplink_addr.h
 * @brief   
 *
 * DAPLink Interface Firmware
 * Copyright (c) 2009-2016, ARM Limited, All Rights Reserved
 * SPDX-License-Identifier: Apache-2.0
 *
 * Licensed under the Apache License, Version 2.0 (the "License"); you may
 * not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 * http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS, WITHOUT
 * WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 */

#ifndef DAPLINK_ADDR_H
#define DAPLINK_ADDR_H

/* Device sizes */

#define DAPLINK_ROM_START               0x00000000
#define DAPLINK_ROM_SIZE                0x00040000

#define DAPLINK_RAM_START               0x20000000
#define DAPLINK_RAM_SIZE                0x0000C000

/* ROM sizes */

#define DAPLINK_ROM_BL_START            0x00000000
#define DAPLINK_ROM_BL_SIZE             0x00007C00

#define DAPLINK_ROM_CONFIG_ADMIN_START  0x00007C00
#define DAPLINK_ROM_CONFIG_ADMIN_SIZE   0x00000400

#define DAPLINK_ROM_IF_START            0x00008000
#define DAPLINK_ROM_IF_SIZE             0x00017000

#define DAPLINK_ROM_CONFIG_USER_START   0x0001F000
#define DAPLINK_ROM_CONFIG_USER_SIZE    0x00021000

/* RAM sizes */

#define DAPLINK_RAM_APP_START           0x20000000
#define DAPLINK_RAM_APP_SIZE            0x00009000

#define DAPLINK_RAM_SHARED_START        0x20009000
#define DAPLINK_RAM_SHARED_SIZE         0x00003000

/* Flash Programming Info */

#define DAPLINK_SECTOR_SIZE             0x00000400
#define DAPLINK_MIN_WRITE_SIZE          0x00000100

/* Current build */

#if defined(DAPLINK_BL)

#define DAPLINK_ROM_APP_START            DAPLINK_ROM_BL_START
#define DAPLINK_ROM_APP_SIZE             DAPLINK_ROM_BL_SIZE
#define DAPLINK_ROM_UPDATE_START         DAPLINK_ROM_IF_START
#define DAPLINK_ROM_UPDATE_SIZE          DAPLINK_ROM_IF_SIZE

#elif defined(DAPLINK_IF)

#define DAPLINK_ROM_APP_START            DAPLINK_ROM_IF_START
#define DAPLINK_ROM_APP_SIZE             DAPLINK_ROM_IF_SIZE
#define DAPLINK_ROM_UPDATE_START         DAPLINK_ROM_BL_START
#define DAPLINK_ROM_UPDATE_SIZE          DAPLINK_ROM_BL_SIZE

#else

#error "Build must be either bootloader or interface"

#endif

#endif

```

c1728p9, 2017-04-26:
Hi @pranavshandilya, what offset does your bootloader to expect the application to start at? If this is out of sync with DAPLink then the image will not be bootable.

pranavshandilya, 2017-04-27:
Hi, @c1728p9  Thanks for your valuable reply..!!
My bootloader expect offset 32KB(0x00008000) but I'm giving 0KB.
Thanks for your help after changing this address, it;s working fine,
Below is the code what I changed.

#include "target_config.h"
```
const char *board_id = "0000";

//  target information
target_cfg_t target_device = {
    .sector_size    = 0x1000,
    // Assume memory is regions are same size. Flash algo should ignore requests
    //  when variable sized sectors exist
    // .sector_cnt = ((.flash_end - .flash_start) / .sector_size);
    .sector_cnt     = (KB(128) / 0x1000),
    .flash_start    = KB(32),
    .flash_end      =KB(128),
    .ram_start      = 0x20000000,
    .ram_end        = 0x2000C000
    /* .flash_algo not needed for bootloader */
};
```
```
const char *board_id = "0000";

// atsam3u2c target information
target_cfg_t target_device = {
    .sector_size    = 0x1000,
    // Assume memory is regions are same size. Flash algo should ignore requests
    //  when variable sized sectors exist
    // .sector_cnt = ((.flash_end - .flash_start) / .sector_size);
    .sector_cnt     = (KB(128) / 0x1000),
    .flash_start    = KB(32),
    .flash_end      =KB(128),
    .ram_start      = 0x20000000,
    .ram_end        = 0x2000C000
    /* .flash_algo not needed for bootloader */
};
```

closed 2017-04-27 by pranavshandilya

c1728p9, 2017-04-27:
I'm glad I could help :smile: