ARMmbed/DAPLink #262: supporting dual cdc (virtual com ports) on PC

Awais174, 2017-04-28:
Is there a way to support multiple mbed com ports via DAPLINK on host PC?
I have tried to update .INF files (C:\Windows\inf\oem47.INF (for mbed composite device) and oem48.INF (for mbed serial device), but driver for 2nd COM port cannot be found. 

Do I need to write my own windows serial driver for dual mbed CDC device? 

**Details:**
I updating DAPLNIK code to create two virtual COM ports along with HID and USB-MSC. I have updated the relevant descriptors and getting following USB device tree view:

        +++++++++++++++++ Device Information ++++++++++++++++++
Device Description       : mbed Composite Device
Device Path              : \\.\usb#vid_0d28&pid_0204#06700000066fff525750877267064158a5a5a5a597969906#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Device ID                : USB\VID_0D28&PID_0204\06700000066FFF525750877267064158A5A5A5A597969906
Hardware IDs             : USB\VID_0D28&PID_0204&REV_0100 USB\VID_0D28&PID_0204
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0064 (GUID_DEVCLASS_USB)
Driver                   : C:\Windows\system32\DRIVERS\mbedComposite_x64.sys (Version: 1.11.0.0  Date: 2009-09-30)
Driver Inf               : C:\Windows\inf\oem47.inf
Legacy BusType           : PNPBus
Class                    : USB
Class GUID               : {36fc9e60-c465-11cf-8056-444553540000} (GUID_DEVCLASS_USB)
Interface GUID           : {a5dcbf10-6530-11d2-901f-00c04fb951ed} (GUID_DEVINTERFACE_USB_DEVICE)
Service                  : mbedComposite
Enumerator               : USB
Location Info            : Port_#0004.Hub_#0003
Container ID             : {1f580a63-03af-57a9-8fbb-3ba95eb2c894}
Manufacturer Info        : mbed
Capabilities             : 0x94 (Removable, UniqueID, SurpriseRemovalOK)
Status                   : 0x0180600A (DN_DRIVER_LOADED, DN_STARTED, DN_DISABLEABLE, DN_REMOVABLE, DN_NT_ENUMERATOR, DN_NT_DRIVER)
Problem Code             : 0
Address                  : 4
Power State              : D0 (supported: D0, D3, wake from D0)
 Child Device 1          : mbed Serial Port
  Device ID              : USB\VID_0D28&PID_0204&MI_01\06700000066FFF525750877267064158A5A5A5A597969906
  Class                  : Ports
  COM-Port               : COM31 (\Device\thcdcacm0)
 Child Device 2          : USB Input Device
  Device ID              : USB\VID_0D28&PID_0204&MI_05\06700000066FFF525750877267064158A5A5A5A597969906
  Class                  : HIDClass
   Child Device 1        : HID-compliant device
    Device ID            : HID\VID_0D28&PID_0204&MI_05\8&14668A70&1&0000
    Class                : HIDClass
 **_Child Device 3          : mbed Serial Port2
  Device ID              : USB\VID_0D28&PID_0204&MI_03\06700000066FFF525750877267064158A5A5A5A597969906
  Problem                : 28 (CM_PROB_FAILED_INSTALL)_**
 Child Device 4          : USB Mass Storage Device
  Device ID              : USB\VID_0D28&PID_0204&MI_00\06700000066FFF525750877267064158A5A5A5A597969906
  Class                  : USB
   Child Device 1        : Disk drive (Disk1)
    Device ID            : USBSTOR\DISK&VEN_MBED&PROD_VFS&REV_0.1\06700000066FFF525750877267064158A5A5A5A597969906&0
    Class                : DiskDrive
     Volume              : \\?\Volume{f81c26c3-1510-11e7-9dcb-b8ca3a81dc59}\
     Kernel Name         : \Device\HarddiskVolume89
     Mountpoint          : F:\

c1728p9, 2017-05-08:
What interface chip are you using? Some of them might not have enough endpoints for an additional CDC port.

Awais174, 2017-05-09:
I am using interface MCU STM32F103, and the desired behavior is HID, MSC and a total of three CDC COM ports. 

In plain DAPLINK, following end-points are configured in usb_config.c:

#define USBD_HID_EP_INTIN           1
#define USBD_HID_EP_INTOUT          1

#define USBD_MSC_EP_BULKIN          2
#define USBD_MSC_EP_BULKOUT         2

#define USBD_CDC_ACM_EP_INTIN           3
#define USBD_CDC_ACM_EP_BULKIN          4
#define USBD_CDC_ACM_EP_BULKOUT         4

And, I suppose, these additions can be made, 

For 2nd COM port:
#define USBD_CDC_2_ACM_EP_INTIN             5 
#define USBD_CDC_2_ACM_EP_BULKIN            6 
#define USBD_CDC_2_ACM_EP_BULKOUT           6 

For 3rd COM port
#define USBD_CDC_3_ACM_EP_INTIN             7 
#define USBD_CDC_3_ACM_EP_BULKIN            8 
#define USBD_CDC_3_ACM_EP_BULKOUT           8 

Reference manual (RM0008) 
STM32F101xx, STM32F102xx, STM32F103xx, STM32F105xx and
STM32F107xx advanced ARM®-based 32-bit MCUs (http://www.st.com/content/ccc/resource/technical/document/reference_manual/59/b9/ba/7f/11/af/43/d5/CD00171190.pdf/files/CD00171190.pdf/jcr:content/translations/en.CD00171190.pdf)

Section 23.2
Configurable number of endpoints from 1 to 8 
Section 23.3 
Up to 16 mono-directional or 8 bidirectional endpoints can be used ...

I am assuming in DAPLINK all the end-points are defined as bi-directional, hence, for STM32F103, I can implement the required configuration utilizing all 8 end-points. 

Can you please correct my understanding if I am wrong.

c1728p9, 2017-05-09:
Hi @Awais174, there is also endpoint 0 which is used for control. Another constraint is the 512 byte USB buffer which must have parts allocated to each endpoint and direction. I believe this buffer was nearly full even without the extra 3 CDC endpoint buffers. Max packet size of some of the endpoints needed to be reduced as seen in [in the stm32 PR here](https://github.com/mbedmicro/DAPLink/pull/257/files#diff-188efbc2c4bbf003b7002bb05de4edd6R315).

Awais174, 2017-05-10:
@c1728p9, so far I have checked that two COM ports do work (one idle, one routing mbed serial printf, and vice versa) using https://github.com/mbedmicro/DAPLink/pull/133 while setting packet size for end-point0 to minimum (8byte).

Thanks a lot for providing the information!! I shall embed and run my implementation again after you merge this pull request with master.

c1728p9, 2017-05-10:
Glad to hear the info helped. Did you need to change the endpoint numbers to get it to work. From your earlier post it looked like you were using 9 endpoints while the stm32f103 only supports 8 endpoints (USB_EP0R-USB_EP7R) which includes the control endpoint.

Awais174, 2017-05-11:
I didn't change existing end-points, but reduced end-point0 packet size to min (8 bytes) and added (3 + 3) - for COM2 and COM3 - end-points in following way:

For 2nd COM port:
#define USBD_CDC_2_ACM_EP_INTIN 5
#define USBD_CDC_2_ACM_EP_BULKIN 6
#define USBD_CDC_2_ACM_EP_BULKOUT 6

For 3rd COM port
#define USBD_CDC_3_ACM_EP_INTIN 7
#define USBD_CDC_3_ACM_EP_BULKIN 8 <<-- not supported by STMF103
#define USBD_CDC_3_ACM_EP_BULKOUT 8 <<-- not supported by STM103

Section 23.3 of STMF103 reference manual states 'configurable end-points from 1-8', and I mistook it to use 'end-point8' in literal sense.  

The device enumeration was successful with PC, but only COM1 and COM2 seem to work for printf serial data (mapped COM2 with user CDC app, and vise versa for COM1); whereas, for COM3, I could only receive garbage data at PC. 

In Daplink, default (COM1) is mapped to USART2, my idea is to send COM2 data to USART3 or USART1. 

Thanks.

c1728p9, 2017-09-25:
Closing this issue. Please reopen @Awais174 if there is more you wanted to discuss.

closed 2017-09-25 by c1728p9