ARMmbed/DAPLink #263: Citrix client not working with mbed DAPLink

DavidWhaleMEF, 2017-05-02:
microbit.org-tickets: 1419, 431, 418, 803

Issues from schools using micro:bits with virtual desktop environments

We are using Xendesktop VDA 7.9 and we can get it recognised when in the maintenance mode as a removable device and in device manager as a portable device, but when it is used normally out of the maintenance mode it is not detected by citrix. We have tried multiple thin clients and multiple micro bits and the usb redirection is enabled.

c1728p9, 2017-05-03:
Hi @DavidWhaleMEF what version of microbit software are you using? Please attach your details.txt.

DavidWhaleMEF, 2017-05-03:
Thanks Russ, this is a teacher that is remote from me, so it is likely to be a pupil board, but I will get them to send me the details.txt file and attach it here when I get it. Thanks very much for your help.

DavidWhaleMEF, 2017-05-03:
Just FYI it might be a bit challenging for the user, because it doesn't work on any of their school systems - bit chicken and egg here (!) So I might have to get them to take the micro:bit home and try it on a non locked down home computer. I'll update you as soon as I have this info though.

DavidWhaleMEF, 2017-05-03:
The Microbit does indeed connect properly when connected to my pc which is outside the citrix environment.

It is one of the free microbits as well and the contents of the Detail.txt file is:

DAPLink Firmware - see https://mbed.com/daplink
Version: 0234
Build:  Oct 12 2015 14:53:22

c1728p9, 2017-05-03:
Hi @DavidWhaleMEF, it looks like that is an older version of DAPLink. I would recommend trying the latest microbit firmware. The this can be found in [DAPLink release 0234](https://github.com/mbedmicro/DAPLink/releases/tag/0243) with the filename 0243_kl26z_microbit_0x8000.hex. It has USB descriptor changes which might fix the problem you are seeing.

DavidWhaleMEF, 2017-05-04:
Thanks Russ. I've written some knew knowledge on our knowledge-base and forwarded this and the hex file to the customer who is using Citrix. I'll update you on what they report back. 

Also we have an old ticket in our system relating to a similar issue on VMWare that we never managed to get to the bottom of, so I have forwarded the solution to that user too.

Thanks very much for being responsive here, Jonny suggested I report the issue over here and the support has been amazing, thanks so much for your help.

c1728p9, 2017-05-04:
I'm glad I could help :smile:

DavidWhaleMEF, 2017-05-04:
Okay, so some feedback from users.

For this one, no change...

1419

We are using Xendesktop VDA 7.9 and we can get it recognised when in the maintenance mode as a removable device and in device manager as a portable device, but when it is used normally out of the maintenance mode it is not detected by citrix. We have tried multiple thin clients and multiple micro bits and the usb redirection is enabled. 

Unfortunately its still doing the same thing where it powers on but doesn't show up and will show up when in maintenance mode. I have also double checked with the network manager and we have all usb devices allowed so it shouldn't need to be whitelisted.

For this vmware one, it fixed the problem, pending further testing...

431, 418, 803

We have installed this on the test unit and this passes through successfully and appears as a microbit in the Virtual Desktop (this is on VMWare Horizon View).
 
We will continue testing and let you know if we find any issues, but suffice to say this is significant progress and may provide 100% of the solution required for this issue.

DavidWhaleMEF, 2017-05-04:
So I think we'll wait to hear back more from the vmware horizon, but that looks good.

Are there any further diagnostics we can get the customer to send us on the citrix Xendesktop VDA 7.9  that would help to narrow this down?

c1728p9, 2017-05-04:
Knowing the OS Xendesktop is running would help

DavidWhaleMEF, 2017-05-05:
Ok thanks, asking customer now for that info and anything else they can give us to describe their setup.

DavidWhaleMEF, 2017-05-09:
microbit.org-ticket: 1419

Xendesktop is running on Windows 2012
Windows 10 is running on the VMs and we are using VDA 7.9
We have no whitelisting rules in place and no rules are installed.

c1728p9, 2017-05-09:
Hi @DavidWhaleMEF, what shows up in device manager for those devices? It is unknown device?

DavidWhaleMEF, 2017-05-09:
Checking...

c1728p9, 2017-09-25:
Hi @DavidWhaleMEF, any updates on this?

DavidWhaleMEF, 2017-09-25:
Unfortunately we lost contact with the customer on this one, the support ticket this end is closed down. Looks like we solved one persons problem but didn't solve the other one.

flit, 2018-11-15:
Closing. Please reopen if this pops up again.

closed 2018-11-15 by flit