ARMmbed/DAPLink #264: about program_page size  in function target_flash_program_page

wayen30, 2017-05-04:
taget_flash.c:

static error_t target_flash_program_page(uint32_t addr, const uint8_t *buf, uint32_t size)
{
    while (size > 0) {
        uint32_t write_size = MIN(size, flash->program_buffer_size);

        // Run flash programming
        if (!swd_flash_syscall_exec(&flash->sys_call_s,
                                    flash->program_page,
                                    addr,
                                    write_size,//flash->program_buffer_size,
                                     ~~~~~~~~~~~~~~
                                        size of program_page should be write_size.                                flash->program_buffer,
                                    0)) {
            return ERROR_WRITE;
        }

c1728p9, 2017-05-04:
Hi @wayen30 I'm not sure what you are asking. Can you elaborate on your question?

wayen30, 2017-05-05:
@c1728p9 
the length of write_data_to_buffer is 'write_size' ,but  the length of program_page is 'flash->program_buffer_size'. I think the length of program_page should be also 'write_size'.

        // Write page to buffer
        if (!swd_write_memory(flash->program_buffer, (uint8_t *)buf, write_size)) {
            return ERROR_ALGO_DATA_SEQ;
        }

        // Run flash programming
        if (!swd_flash_syscall_exec(&flash->sys_call_s,
                                    flash->program_page,
                                    addr,
                                    flash->program_buffer_size,
                                    flash->program_buffer,
                                    0))