ARMmbed/DAPLink #267: Assert at msc\usbd_msc.c line 1081

KariHaapalehto, 2017-05-12:
Hi!

Assert at usb has been several times, if there is a lot of usb traffic.

File: ..\..\..\source\usb\msc\usbd_msc.c
Line: 1081
Source: Application

DAPLink 043 version was used.

[ASSERT.TXT](https://github.com/mbedmicro/DAPLink/files/995707/ASSERT.TXT)
[DETAILS.TXT](https://github.com/mbedmicro/DAPLink/files/995708/DETAILS.TXT)

SeppoTakalo, 2017-05-12:
CC: @c1728p9 Do you know who can help to investigate this?

c1728p9, 2017-05-12:
Thanks for attaching all the relevant information @KariHaapalehto. @SeppoTakalo, yep, its me who can help with DAPLink issues :smile:. I think this issue has been fixed by the change https://github.com/mbedmicro/DAPLink/commit/7e984f8d3e16c668a6999e1d341d33e84518c595. Unfortunately, there hasn't been a release with this change yet. I created a K64F build from master (which has this fix) and attached it. Please give it a try and let me know if this fixes your problem.
[0243+_k20dx_frdmk64f_0x5000_2f498feb.zip](https://github.com/mbedmicro/DAPLink/files/997116/0243._k20dx_frdmk64f_0x5000_2f498feb.zip)

KariHaapalehto, 2017-05-16:
Hi!

The fix didn’t help. Assert is still seen, now at line 1089.

Assert
File: ..\..\..\source\usb\msc\usbd_msc.c
Line: 1089
Source: Application

Br,
-Kari-

c1728p9, 2017-05-16:
Hello @KariHaapalehto, which OS and which version of that OS are you using? Also are you using a VM and if so using program? Finally, do you have the K64F connected directly to your computer or is it going through a hub? If going through a hub, which hub are you using?

KariHaapalehto, 2017-05-17:
Hi!

I'm using windows 10 and VirtualBox. At virtual box I have 4 K64F connected to minicom terminal sw to see traces . Since I don't have 4 usb port's available at my laptop, they have been connected through hub or directly to docking station and I have seen usb reset in both cases.

c1728p9, 2017-05-18:
This assert indicates that the host sent an OUT packet over the mass storage interface when the device was expecting to send an IN packet first to report the status of the previous command. This indicates that the problem is likely on the host computer, or possibly the hub. If you can capture the traffic with a USB analyzer then the source of the problem can be determined.

In addition, what kind of hub are you using? If it isn't a MTT hub you'll see problems on Kinetis devices, so make sure you are using one of these. 

Also, there are known problems with the VirtualBox USB drivers which can cause packets to get dropped. I would recommend trying this same test on a native Windows 10 machine to see if you have the same problem. At the very least, don't use VirtualBox, use something like VMWare.

KariHaapalehto, 2017-05-22:
Seems like the Windows 10 is causing this, since I saw the same error with putty on windows side. But when I moved my test set-up to native ubuntu machine, I haven't see this problem anymore.  This will help me for now, but how about all other developes, who will use windows 10?

c1728p9, 2017-05-22:
Hi @KariHaapalehto, it is unlikely this is a problem with Windows 10. Was Windows 10 running inside VirtualBox, or were you running something else? Please provide steps to reproduce this problem so I can try it locally. Also please provide me the model number of the USB hub you were using.

c1728p9, 2017-07-22:
Hi @KariHaapalehto any updates on this?

KariHaapalehto, 2017-07-25:
The hub is D-Link's DUB-H7 and I had 4 K64F connected to it and they were pinging each other.

c1728p9, 2017-09-25:
Closing this issue since it has been fixed. Please re-open if this is not the case.

closed 2017-09-25 by c1728p9

bmcdonnell-ionx, 2018-05-09:
I am experiencing this issue.

```
Assert
File: ..\..\..\source\usb\mss\uóbl_msc.c
Line: 1081
Source: Application
```

Host machine:
- Lenovo T470 (Type 20HD, 20HE) Laptop (ThinkPad)
- Windows 10 Enterprise x64
- *Not* using any VirtualBox / VM software.
- I usually use a USB port on the back of the [Ultra Dock](https://support.lenovo.com/us/en/solutions/pd028010).

Target: NXP LPC4088 on [EA LPC4088QSB](https://www.embeddedartists.com/products/boards/lpc4088_qsb.php).

[ASSERT.TXT](https://github.com/ARMmbed/DAPLink/files/1988779/ASSERT.TXT)
[DETAILS.TXT](https://github.com/ARMmbed/DAPLink/files/1988780/DETAILS.TXT)
[MBED.HTM.txt](https://github.com/ARMmbed/DAPLink/files/1988782/MBED.HTM.txt)

c1728p9, 2018-05-10:
Hi @bmcdonnell-ionx, thanks for the detailed error report. The version shown in details.txt you have attached is 0241 which is before this fix was made (fix was made in 0244).

You can get the latest firmware here:
https://armmbed.github.io/DAPLink/

Let me know if you see any problems with the new firmware.

bmcdonnell-ionx, 2018-05-10:
@c1728p9,

Thanks for the timely and informative reply.

I had previously downloaded and loaded what I thought was the latest firmware (0241), provided by EA [here](https://os.mbed.com/users/embeddedartists/notebook/lpc4088-updating-the-cmsis-dap-interface/). The [product page](http://www.embeddedartists.com/products/boards/lpc4088_qsb.php) says, "[All documentation](https://os.mbed.com/users/embeddedartists/notebook/lpc4088-quickstart-board/) has been put on the mbed.org site." That page links to the previous, with the outdated firmware. I had not seen the page you linked before.

I left [a comment requesting](https://os.mbed.com/users/embeddedartists/notebook/lpc4088-updating-the-cmsis-dap-interface/#c30130) that they update the documentation. Feel free to nudge them to do the same.