ARMmbed/DAPLink #268: Query about 'stuck in maintenance mode'

DavidWhaleMEF, 2017-05-15:
microbit.org-ticket: 1789

We occasionally get queries from micro:bit users who say that their micro:bit is stuck in maintenance mode, a typical query is below FYI...this is from a pupil board with the standard firmware preloaded.

_Did not work on a micro:bit stuck in maintenance mode; after the .hex file was transferred, there was a lot of flashing, the drive reconnected and nothing else happened. DETAILS.TXT still showed the old version._

It would be good to understand the causes for how an mbed board in particular can get stuck in maintenance mode. Is this due to a faulty device, or is it a software configuration error?

DavidWhaleMEF, 2017-05-15:
@jaustin Do we have any specific knowledge as to how a mbed board/micro:bit gets stuck in maintenance mode?

c1728p9, 2017-07-22:
Hi @DavidWhaleMEF, DAPLink can get stuck in maintenance mode if the interface firmware gets corrupted in an update. This can happen if a user unintentionally loads a binary for the nrf51 file while in maintenance mode. Probably the best way to determine the cause is to read out the flash of the KL26 with a debugger. That way you can tell where the corruption is, or if the user loaded an invalid image.

c1728p9, 2017-09-25:
Hi @DavidWhaleMEF, please re-open if you have further questions.

closed 2017-09-25 by c1728p9

DavidWhaleMEF, 2017-09-25:
Ok thanks. We have this tagged in our system for addition to the knowledge-base, so the info you provided has been helpful for writing that article, thanks!