ARMmbed/DAPLink #269: Using daplink with Android over USB

DavidWhaleMEF, 2017-05-15:
microbit.org-ticket: 1558

Users are struggling to get Android to detect the micro:bit as a removable drive over USB. Some information below. Is this something we would expect to work, do you know? Thanks!

I would feel much happier using USB from a tablet and have been running some tests with USB OTG. So far I haven't been able to talk to the micro:bit, presumably because there is no software support for OTG in the micro:bit. Would you consider making the micro:bit compatible with USB OTG?

platform: Samsung Tab 4

Using USB OTG File Manager by Top Photo Inc I can read and write files to a memory stick using an OTG adaptor cable. 

When I connect the micro:bit instead of the memory stick the file manger displays the device as an "mbed", but doesn't show any files. I've tried copying a compiled .hex file to the mbed but it doesn't transfer.

Original firmware was:
DAPLink Firmware - see https://mbed.com/daplink
Version: 0234
Build:   Oct 12 2015 14:53:22

Updated firmware to: 
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9900000031864e45006d100b0000002a0000000097969901
HIC ID: 97969901
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0243
Git SHA: b403a07e3696cee1e116d44cbdd64446e056ce38
Local Mods: 0
USB Interfaces: MSD, CDC, HID
Interface CRC: 0x14256f44
Remount count: 0

Now it responds as follows:

It's behaviour with the tablet over USB OTG is now slightly different in that the micro:bit is displayed as an ARM mass storage device rather than an mbed. However I still can't read or write files to it using USB OTG File Manager by Top Photo Inc .

Also the yellow LED only comes on briefly when I connect USB whereas previously the yellow LED was on continuously. This point is probably insignificant, but I thought I'd note it just in case it's relevant.

Just a thought - my understanding is that USB OTG only supports FAT32  and NTFS file formats. I don't know what format(s) the micro:bit supports, but could this be the problem?

I've reformatted my memory stick to FAT16 and can read and write files to it from the tablet, even though Android isn't supposed to support FAT16. Maybe the file Manger I'm using does support this format, although it doesn't say so in the spec.

I've been unable to reformat my memory stick to FAT12.

DavidWhaleMEF, 2017-05-22:
@c1728p9 Hi Russ - do you or the team have any ideas whether this should work or not out of the box?

Thanks

c1728p9, 2017-05-22:
Hi @DavidWhaleMEF, I'm not familiar with the differences between normal USB and USB OTG so I don't know for sure. If OTG is compatible with regular USB devices then this should work. What kind of setup to you have that allows a micro:bit to be connected to a phone?

DavidWhaleMEF, 2017-05-22:
Hi Russ,

Typically we would use a lead a bit like this: http://www.maplin.co.uk/p/maplin-tech-on-the-go-micro-usb-to-female-usb-n36qg and then plug the normal micro:bit short USB lead into the female socket, and the micro goes into the bottom of the Android phone.

I have seen only one report from a user that they have got this to work, but I have seen reports from other users that they can't get it to recognise the drive (mostly captured above).

I've also failed to get this to work on my HP Slate 10 (Android) tablet with an OTG lead - nothing new appears in the device manager when I plug it in.

At this stage, I was just interested in us trying to work out if this *should* work - and perhaps beyond that, it might be that we need to understand if there are specific requirements on a minimum Android version or something that would be required to make it work.

I do still have my HP Slate 10 device and OTG lead here and could perform some additional diagnostics for you if that would help, but I'd need to know what to look for.

flit, 2017-05-23:
USB OTG is a set of signaling protocols that use the USB ID pin to:
- determine whether a device is attached (Attach Detection Protocol, ADP)
- control power to an attached device (Session Request Protocol, SRP)
- negotiate host/device role swapping (Host Negotiation Protocol, HNP)

For devices that don't support the above protocols, host/device role is determined by the cable. Most devices do _not_ support these protocols, or even connect the ID pin. Beyond that, it's regular USB 2.0, including USB mass storage.

Thus, the comment about OTG supporting only FAT32 or NTFS does not really make sense. It's the Android app that determines which filesystems are supported.

Try putting a hub between the Android device and micro:bit. That should force normal host/device roles, in case that was an issue.

See the [wikipedia article](https://en.wikipedia.org/wiki/USB_On-The-Go) for an overview.

DavidWhaleMEF, 2017-05-24:
Thanks. Will get user to try that, and report back what we find. Thanks for your help.

DavidWhaleMEF, 2017-05-29:
Customer reported that your advice made no difference at all, unfortunately:

I've tested the following configuration, which sadly did not work. 

Tablet > USB OTG Cable > USB Hub > micro:bit

The behaviour with the hub is exactly the same as it is with just the OTG cable, i.e. the file manager simply reports the micro:bit as an ARM device and isn't possible to upload a .hex file.

flit, 2017-05-30:
Thanks for trying, sorry it didn't work. My guess is that it's actually an issue with the file manager app.

microbit-mark, 2017-09-25:
I've had another report asking why the micro:bit won't show up as a device when attached via USB to Android device. Is there a definitive answer on this? I've also tried with a couple of other free USB OTG file manager apps, but I don't see a connection or get a yellow LED.

microbit-carlos, 2017-09-25:
I tried connecting it to my android  (6.0.1) device with an OTG cable and the OS won't mount the drive at all (while a normal FAT32 USB Flash drive works).
Not only that, but the operating system shows me a notification saying that the drive is corrupted, and asks to format it to solve the issue.
Also tried with an app claiming to be FAT12 compatible (Total commander with USB drive pluging) and it crashes when trying to access the DAPLink MSD.

ensonic, 2017-11-24:
Same here. I tested a normal flash drive witha micro-usb adapter on a nexus 7 tablet and via usb-c adapter on a pixel-c and those work. But my micro:bit or calliope mini don't. The devices get power, so the light up for me, but don't show up on the android device like the actual flash drive.

rsrini7, 2018-01-01:
I am also looking for USB OTG support for micro:bit. Just purchased, will try once i received.

microbit-mark, 2018-01-19:
microbit.org-ticket: 4196

User reports:

 Procedure: Connect micro:bit to an Android device with USB-host support (USB
  OTG).

  Expected result: micro:bit shows up as an external media device, so
  HEX files can be dropped there.

  Actual result: an Android notification appears reporting a damaged
  external media card, and offers to reformat the card.

User use Android 6 and Android 7 version in different tablets and cellphones for testing.

DavidWhaleMEF, 2018-01-19:
Hi @c1728p9 is there any knowledge about this ticket from the DAPLink team yet, do you know? It's a common use-case we see people trying to apply to micro:bit, and unfortunately there seems to be no traction from any angles on this at the moment. It would be nice to get to the bottom of this one, as it could enable more use of micro:bit in pop-up clubs and libraries etc where tablets and phones are commonly used for workshops. Thanks!

P.S. understanding the problem here is probably more important (at the moment) than actually fixing it. If we can get some clarity on understanding what the cause of the problem is, we may be able to then find the right help in the community to fix it.

DavidWhaleMEF, 2018-01-19:
@c1728p9 We have a number of support tickets from users now, all of which we can access for testing and further observation. @microbit-mark can link you up with those users if that would help?

DavidWhaleMEF, 2018-01-19:
Here is evidence that it did work (once), via @martinohanlon (note the date, and the vintage of the phone)

https://twitter.com/martinohanlon/status/673906085040947200

[image]

DavidWhaleMEF, 2018-01-19:
Although, I have failed to get USB flashing from any of my Android phones to work at all.

cederom, 2018-01-21:
Here is initial analysis based on the provided above capture files:
* Packet 560 is the last USB descriptor read.
* Packet 575 is the "Get Max LUN" request starting SCSI/UMS operations, after that we have periodic "Test Unit Ready" and "Mode Sense" requests.
* Packet 619 is "Read Capacity" with response of Returned Logical Block Address 131199, Logical Block Length (Bytes) 512, Total capacity 64.65 MB.
* Packet 670 "Prevent Allow Medium Removal" command marked as Failed.
* Packet 762 is the first read from drive LBA=0 Length=8Blocks. This returns MBR.
* Following reads are in packets 1154 (LBA=0,Len=32Blocks), 2423 (LBA=0,Len=32Blocks) that also returns MBR.
* Then packet 3968 shows read LBA=131072 Len=8Blocks that returns all zeros!
* Then packet 4028 shows read LBA=0 Len=240Blocks that returns MBR.
* Then packet 13485 shows read LBA=240 Len=16Blocks that returns all zeros!
* Five more reads like that and Android shows "Drive Corrupt" warning.

It looks like Android is searching for something on locations LBA=240 and LBA=131072 and expects something else than zeros. LBA=240 could be start of the filesystem (FAT?). LBA=131072 could be end of drive (backup GPT?).

This looks like Andoid Automounter issue. DAPLink does not provide valid partition table. But that does not cause problems on Linux (that Android runs on).

Another possibility is that DAPLink on Microbit crashes during those initial reads. I will check other DAPLink boards where I can debug the DAPLink itself whether they have the same problem on Android..

cederom, 2018-01-21:
It is also worth noting that on several other older phones the USB Mass Storage over OTG did not work for me at all. Make sure any other USB Mass Storage drive works on your device if you have problems with Microbit/DAPLink.. because your device may simply not support USB OTG Mass Storage :-)

cederom, 2018-01-22:
All other boards have the same "MBED USB drive is broken" problem on Android, using DAPLink 244 on K64F, nRF51, NCS36510, R7S, LPC4088...

cederom, 2018-01-23:
I did an experiment:
* dumped with DD byte-by-byte image of the Microbit DAPLink UMS to a file:
`dd if=/dev/disk2 of=20180123-microbit_ums_dd_dump.bin`
* then restored that image to a USB Mass Storage Drive (SD card reader):
`dd if=20180123-microbit_ums_dd_dump.bin of=/dev/disk2`

It showed broken filesystem on Android. Maybe VFS implementation issue and Android Automounter can't handle that. Also can work on that image to search for fix without code rewrite..

cederom, 2018-01-24:
I will note bolow some findings about MBR/VBR/FAT that may come handy here:
* DAPLink contains `55 AA` signature at `0x1FE` address, that according to [1] means bootloader code is present, but there is no bootloader code. We should probably remove that signature.

[1] https://en.wikipedia.org/wiki/Volume_boot_record

cederom, 2018-02-12:
Just to make sure all is aligned well with Android I have created an Issue Ticket [1] and asked Google on Specification / Requirements for USB OTG Mass Storage drives. Probably FAT VBR is not enough and we would have to rewrite some of the DAPLink VFS to make it work on older Android versions such as 5,6,7,8..

[1] https://issuetracker.google.com/issues/73130696

cederom, 2018-02-24:
Good news, problem solved, patches already sent to upstream PR https://github.com/ARMmbed/DAPLink/pull/401! :-)

Tested and works on:
+ Nexus 5X Android 8.1.0
+ Sony Xperia Z1 Android 5.1.1
+ Sony Xperia M5 Android 6.0
+ Samsung Galaxy S2 Android 2.3.3

You can track log of work at [fix] and try out the Test Microbit Firmware [testfw].

Have a nice weekend! :-)

[fix] https://github.com/microbitedupl/shared/tree/master/fix/2018-microbit-daplink-android-mount_problem
[testfw] https://github.com/microbitedupl/shared/blob/master/fix/2018-microbit-daplink-android-mount_problem/firmware/kl26z_microbit_if_crc_legacy_0x8000-Android_OTG_test-20180224.bin.7z?raw=true

DavidWhaleMEF, 2018-04-19:
@cederom I was just referring some of my contacts to this ticket, and I realised (at least) three things

1) This is an epic piece of diagnostics work!
2) We never actually said 'thanks'
3) Help from Google engineers as well - double epic! 

So, from all of us at Micro:bit Foundation, from all of our users who have been hoping to use this feature, a BIG THANK YOU!!!!

This is a shining example of excellent collaboration between disparate teams to solve a complex problem.

cc @jaustin @microbit-mark @microbit-carlos

cederom, 2018-04-23:
Thank you @DavidWhaleMEF, glad I could help! :-) I am also working on big scale launch of Microbit in Polska :-)

0xc0170, 2019-12-11:
Thanks @cederom for fixing this. I'll close as resolved.

closed 2019-12-11 by 0xc0170