ARMmbed/DAPLink #270: interrupting debugging causes unexpected data to serial port

jupe, 2017-05-18:
## Steps to reproduce
1) run pyocd-gdbserver with K64F (dappling v0243)
2) `(gdb) continue`
3) open serial port connetion
    Now serial communication seems to work as expected
3) send SIGINT to interrupt debugging:
    ```
    (gdb) <ctr+c>
    Program received signal SIGINT, Interrupt.
    0x20000014 in ?? ()
    ```
4) strange data appears to serial port:
    ```
    $T0207:00000000;0d:281e0020;0e:11890000;0f:22890000;#1a$c04080c0c000000000008
    00000000000c04080c0e0010100800b002000000000140000206c0c0020dc0c002000000000af680000281e0020118900002289000000000041d0ff022028
    1e002000000000020000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
    0000000000000000000000000000000000000000000000000000#6f$ff#cc$ff#cc$0000000090eb0000b9eb000055eb0000abeb0000ff00000066eb00001
    feb000032eb00007eeb000076eb0000b2eb000086eb0000a1eb000044eb000001000000#66$l<?xml version="1.0"?><!DOCTYPE feature SYSTEM "th
    reads.dtd"><threads><thread core="0" id="1">Thread mode</thread></threads>#aa
```

It looks like some buffer overflows etc..
Might also related to https://github.com/mbedmicro/pyOCD/issues/309.

jupe, 2017-05-18:
This might be actually invalid defect. I'll try with different way and close if I cannot reproduce.

flit, 2017-05-18:
The data you show is the gdb server protocol stream that is being sent to/from pyocd-gdbserver and gdb.

jupe, 2017-05-19:
Yep, this was my mistake - test environment I did use leaks data from gdb channel to serial port.

closed 2017-05-19 by jupe