ARMmbed/DAPLink #271: DAPLink (0x243 version) has error on Ubuntu 16.04 64bit

iotvietmember, 2017-05-25:
Hi,

I cloned this repo and build with Keil 4. 
I modified default setting in setting_rom.c. I set config parameter from 0 to 1 when system start.
Everything is successful.

On Windows, this board is work fine. But it is an error with MSD feature on Ubuntu 16.4 64bit.

On Ubuntu, when I copy a HEX file to DAPLINK partition, MSD led does not blink as on Windows and download progress is not occur

Thanks!

c1728p9, 2017-05-25:
Hi @iotvietmember, what config parameter did you change from 0 to 1? Also, are you running Ubuntu 16.04 in a virtual machine and if so which one? Finally, does MSD show fail.txt after programming fails on Ubuntu?

iotvietmember, 2017-05-25:
Hi @c1728p9 . Thanks for your reply.

I modified `setting_rom.c` as the below code.
```
// Configuration defaults in flash
static const cfg_setting_t config_default = {
    .auto_rst = 1,					//default: 0			
    .automation_allowed = 1,			//default: 0
    .overflow_detect = 1,				//default: 0	
};
```
But I think this thing does not made error.

Beside, I also modified  file `usbd_user_cdc_acm.c` as below code.
```
int32_t USBD_CDC_ACM_PortSetLineCoding(CDC_LINE_CODING *line_coding)
{
    UART_Config.Baudrate    = line_coding->dwDTERate;
    UART_Config.DataBits    = (UART_DataBits) line_coding->bDataBits;
    UART_Config.Parity      = (UART_Parity)   line_coding->bParityType;
    UART_Config.StopBits    = (UART_StopBits) line_coding->bCharFormat;
    UART_Config.FlowControl = UART_FLOW_CONTROL_RTS_CTS;        //default: UART_Config.FlowControl = UART_FLOW_CONTROL_NONE;
    return uart_set_configuration(&UART_Config);
}
```
And **I HAVE DETERMINED THAT THIS IS THE CAUSE OF THIS ERROR**.
But on windows, every thing is ok. 
On Ubuntu, set  `UART_FLOW_CONTROL_RTS_CTS` for CDC_ACM on **LPC11U35** made an error

iotvietmember, 2017-05-25:
@c1728p9 : 

I found the below code in the `uart.c` file on LPC11U35 chip
```
void uart_enable_flow_control(bool enabled)
{
    // Flow control not implemented for this platform
}
```
I think that LPC11U35 support UART Hardware flow control (RTS/CTS). 
I do not understand the above code. 
Would you like explain it?
Thanks.

c1728p9, 2017-05-25:
Good to hear that you have found the cause of the error. I'm surprised this is causing problems on Ubuntu.

Also, unrelated, but with the latest lpc11u35 code you can change auto_rst, automation_allowed and overflow_detect at runtime by creating .cfg files. Info on this can be found:
[how to enable automation](https://github.com/mbedmicro/DAPLink/blob/master/docs/ENABLE_AUTOMATION.md)
[mass storage commands](https://github.com/mbedmicro/DAPLink/blob/master/docs/MSD_COMMANDS.md)

c1728p9, 2017-05-25:
The function ```uart_enable_flow_control``` is available so boards can turn flow control on or off. For example the ncs36510 turns it off [here](https://github.com/mbedmicro/DAPLink/blob/master/source/board/ncs36510rf.c#L23). It looks like the comment on the LPC11U35's version of ```uart_enable_flow_control``` is wrong since it appears that flow control is implemented [here](https://github.com/mbedmicro/DAPLink/blob/master/source/hic_hal/nxp/lpc11u35/uart.c#L150).

iotvietmember, 2017-05-25:
@c1728p9 
Yes, UART Hardware flow control on LPC11U35 (connect to nRF51822) is ok on Windows.
Perhaps I will found reason soon. Please wait for me

iotvietmember, 2017-05-25:
@c1728p9 
Sorry for wasting your time. But actually I'm not sure what the main cause of the error.

1. When I set `auto_rst=1, automation_allowmode=1, overflow_detect=1 in code` and set `UART hardware flow control with RTS/CTS`:
1.1. On Ubuntu 16.04 LTS 64bit: I can not copy HEX file to DAPLINK partition over MSD. Progress is crash.
1.2. **On Raspberry Pi 3 (Raspbian OS - based on Debian - an Linux distribution): Everything is OK**
1.3. **On Window OS: Everything is OK**

Would you like help me, please!

c1728p9, 2017-05-25:
Hi @iotvietmember, how can you tell there is a crash? Do you have a debugger connected?  DAPLink should reboot after a crash.

iotvietmember, 2017-05-26:
Hi @c1728p9 , sorry for reply late. 
Crash: when I copy HEX file to DAPLINK partition, the progress is not occur and MSD led does not blink. I try to copy HEX file one time, the progress is still not successful. And from that time, I can not copy any the HEX file to any DAPLINK board.
I reboot DAPLInk after crash but the results is the same. 

My board consits of 1 nRF51822 and 1 LPC11U35.

iotvietmember, 2017-05-26:
@c1728p9 

When I set `auto_rst=1, automation_allowmode=1, overflow_detect=1` in code and set `UART_Config.FlowControl = UART_FLOW_CONTROL_NONE`:
Everything is ok on Ubuntu 16.04 64 bit.

Why do I set Uart Hardware Flow Control made an error for download HEX file to DAPLInk partition?

iotvietmember, 2017-05-26:
@c1728p9 

I think Ubuntu has error with usb_cdc_acm driver when device set Uart hardware flowcontrol.

When copy HEX file to DAPLINK partition, an error occur.  I use `dmesg` command and the results is as the below image.

[image]

After that, I continue to use `dmesg` one more time and the results is as the below image.
[image]

Seem to be ubuntu met an error when try to set dtr/rts.

c1728p9, 2017-05-26:
Hi @iotvietmember, it looks like the lpc11u35 is getting stuck trying to reset the uart. The reset function waits for all the bytes to be sent before returning, but since flow control is on then it never happens if the target doesn't have CTS asserted. I create #273 to fix this. Please give it a try and let me know if it fixes your problem.

iotvietmember, 2017-05-27:
Hi @c1728p9 . I tried to #273 . I think that that solution fixed my problem with lpc11u35 flow control on Ubuntu.
I try to copy HEX file to DAPLINK partition on Ubuntu 16.04 LTS 64 bit and KaliLinux 64 bit. ** EVERY THING IS OK**

But I have not tried on Ubuntu 32 bit.

Many thanks to you.

c1728p9, 2017-05-31:
Hi @iotvietmember, this problem wasn't Ubuntu specific. It can occur even on windows if baudrate is changed after sending data. #273 should fix the lockup for all platforms.

closed 2017-05-31 by iotvietmember