ARMmbed/DAPLink #277: Daplink can't be detected in Keil 5.18

wayen30, 2017-06-21:
Daplink can't be detected in Keil 5.18, but it's ok in Keil 4.70a.
After copy CMSIS_DAP.dll from Keil 4.70a to Keil 5.18，the daplink in keil5.18 is ok .
Why?

c1728p9, 2017-06-22:
Which version of DAPLink is on your board?

wayen30, 2017-06-23:
Keil Debug show "DAPLink CMSIS-DAP""Firmware Version: 1.0"

I build DAPLink firmware with source code. 
the latest commit in source code is:
=================================================================
commit 2f498feb503c4396822ef00f6fbf8c5289dfa9c0
Author: Russ Butler <russ.butler@arm.com>
Date:   Wed Mar 29 16:47:02 2017 -0500

    lpc11u35 - add flash driver

    Add an internal flash driver for the lpc11u35. This enables this
    interface to store settings.

c1728p9, 2017-07-10:
I'm using 5.23 and I don't have any problems detecting DAPLink.  Can you attach ```details.txt``` from the mass storage drive of your board? Also if you could respond with the version of Keil tools it would help. For me, the versions I'm using are:
```
Tool Version Numbers:
Toolchain:        MDK-ARM Professional  Version: 5.23
Toolchain Path:    C:\Keil_v5\ARM\ARMCC\Bin
C Compiler:         Armcc.exe        V5.06 (build 20)
Assembler:          Armasm.exe        V5.06 (build 20)
Linker/Locator:     ArmLink.exe        V5.06 (build 20)
Library Manager:    ArmAr.exe        V5.06 (build 20)
Hex Converter:      FromElf.exe        V5.06 (build 20)
CPU DLL:               SARMCM3.DLL          V5.23
Dialog DLL:         DARMCM1.DLL          V1.17.0.0
Target DLL:             Segger\JL2CM3.dll      V2.99.23.0
Dialog DLL:         TARMCM1.DLL          V1.12.0.0
```

flit, 2017-07-11:
@wayen30 On which board are you running DAPLink? Also, does the latest MDK version 5.23 work as expected?

c1728p9, 2017-07-22:
@wayen30 can you provide the Keil Tool version numbers?

wayen30, 2017-07-24:
@flit
Daplink is based on lpc11u35_tiny_if. The target board is a soc based on Cortex-M0.
I do not have MDK version 5.23, so I have not tried it.

@c1728p9
Tool Version Numbers:
Toolchain:        MDK-ARM Professional  Version: 5.18
Toolchain Path:    C:\Keil_v5\ARM\ARMCC\Bin
C Compiler:         Armcc.exe        V5.06 update 1 (build 61)
Assembler:          Armasm.exe        V5.06 update 1 (build 61)
Linker/Locator:     ArmLink.exe        V5.06 update 1 (build 61)
Library Manager:    ArmAr.exe        V5.06 update 1 (build 61)
Hex Converter:      FromElf.exe        V5.06  update 1 (build 61)
CPU DLL:               SARMCM3.DLL          V5.18
Dialog DLL:         DARMCM1.DLL          V1.12.0.0
Target DLL:            CMSIS_AGDI.dll      V1.25.7.0
Dialog DLL:         TARMCM1.DLL          V1.9.0.0

wayen30, 2017-07-24:
@c1728p9

My Daplink is modified on the basis of lpc11u35_tiny_if.

details.txt  of Daplink.
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 560100001c8c8c8600000000000000000000000097969902
HIC ID: 97969902
Auto Reset: 0
Automation allowed: 0
Overflow detection: 0
Daplink Mode: Interface
Interface Version: 0243
Git SHA: 2f498feb503c4396822ef00f6fbf8c5289dfa9c0
Local Mods: 1
USB Interfaces: MSD, CDC, HID
Interface CRC: 0x72abe328
Remount count: 0

closed 2021-06-08 by mathias-arm