ARMmbed/DAPLink #278: DAPLink compilation failed

neomissing, 2017-06-23:
Program Size: Code=39740 RO-data=5360 RW-data=588 ZI-data=13088  
FromELF: creating hex file...
.\build\kl27z_kl27z_if.hex: Error: Q0108E: Could not create output file '.\build\kl27z_kl27z_if.hex': Permission denied
Finished: 0 information, 0 warning and 1 error messages.
After Build - User command #1: ..\..\..\tools\post_build_script.bat C:\Users\lqy\Desktop\DAPLink\projectfiles\uvision\kl27z_kl27z_if\build\kl27z_kl27z_if d:\Keil_v5\ARM
C:\Users\lqy\Desktop\DAPLink\projectfiles\uvision\kl27z_kl27z_if>fromelf --bin C:\Users\lqy\Desktop\DAPLink\projectfiles\uvision\kl27z_kl27z_if\build\kl27z_kl27z_if.axf -o C:\Users\lqy\Desktop\DAPLink\projectfiles\uvision\kl27z_kl27z_if\build\kl27z_kl27z_if.bin 
C:\Users\lqy\Desktop\DAPLink\projectfiles\uvision\kl27z_kl27z_if>fromelf --i32 C:\Users\lqy\Desktop\DAPLink\projectfiles\uvision\kl27z_kl27z_if\build\kl27z_kl27z_if.axf -o C:\Users\lqy\Desktop\DAPLink\projectfiles\uvision\kl27z_kl27z_if\build\kl27z_kl27z_if.hex 
C:\Users\lqy\Desktop\DAPLink\projectfiles\uvision\kl27z_kl27z_if>python ..\..\..\tools\post_compute_crc.py C:\Users\lqy\Desktop\DAPLink\projectfiles\uvision\kl27z_kl27z_if\build\kl27z_kl27z_if.hex C:\Users\lqy\Desktop\DAPLink\projectfiles\uvision\kl27z_kl27z_if\build\kl27z_kl27z_if_crc 
Traceback (most recent call last):
  File "..\..\..\tools\post_compute_crc.py", line 137, in <module>
    main()
  File "..\..\..\tools\post_compute_crc.py", line 58, in main
    new_hex_file.fromfile(input_file, format='hex')
  File "C:\Python27\lib\site-packages\intelhex\__init__.py", line 246, in loadfile
    self.loadhex(fobj)
  File "C:\Python27\lib\site-packages\intelhex\__init__.py", line 196, in loadhex
    fobj = open(fobj, "r")
IOError: [Errno 13] Permission denied: 'C:\\Users\\lqy\\Desktop\\DAPLink\\projectfiles\\uvision\\kl27z_kl27z_if\\build\\kl27z_kl27z_if.hex'
".\build\kl27z_kl27z_if.axf" - 1 Error(s), 1 Warning(s).
Target not created.
Build Time Elapsed:  00:00:27

wayen30, 2017-07-03:
You can try running keil with administrator privileges

c1728p9, 2017-07-22:
Hi @neomissing, are you still having this problem? If so could you provide steps to reproduce this from a fresh checkout of DAPLink?

neomissing, 2017-07-23:
I use the auto-generated method, but that is the failure,    progen generate -t uvision

c1728p9, 2017-07-27:
Does the file or directory `.\build\kl27z_kl27z_if.hex` exist?

c1728p9, 2017-09-25:
Please re-open this issue if you have more information or need additional help.

closed 2017-09-25 by c1728p9