ARMmbed/DAPLink #281: DAPLink boot for K20  and app for KL26 not work

neomissing, 2017-07-07:
I try to compile DAPLink BOOT, found that the firmware can not work after programming, compile CMSIS DAP BOOT, can work properly, DAPLink APP does not work

c1728p9, 2017-07-22:
Hi @neomissing, what APP firmware are you trying to use and for what board? Some versions of the CMSIS-DAP bootloader have an application offset of 0x5000, while all versions of the DAPLink bootloader have an application offset of 0x8000.

neomissing, 2017-07-23:
I compiled DAPLink K20 boot, but no response to the K20 programming, I found that many people have this problem, I try to compile CMSIS DAP K20 boot, programmed to K20 can work, I also compiled the same programming DAPLink APP, found It is not working

c1728p9, 2017-07-27:
@neomissing can you attach both bootloader and the application you are loading so I can take a look?

c1728p9, 2017-09-25:
Please re-open this issue if you have more information or need additional help.

closed 2017-09-25 by c1728p9